riscvarchive/riscv-code-size-reduction #2: Use of *cause for uOP sequencing, restart and debug.

David-Horner, 2020-09-14:
#### Huawei Custom Extension/riscv_uop_handling.rst
proposes a solution to the uOPcode sequencing and management of expanded opcodes.

The proposal uses upper bits of *cause to provide a "restart" value.

There is no formal proposal yet but to the extent it may model after Huawei's workproduct, these are my concerns:

1) Currently the Fast-Interrupt TG proposes using *cause for aggregated interrupt information.
There is a potential conflict if both "standards" use the same upper bits.
Better for neither proposal to use the base *cause, but rather an alternate newly dedicated csr for these purposes.
The V-extension intentionally avoids extending existing csr except as previously left reserved for such future stndard usage.

2) Ideally, restart/suspend information should be opaque with directives and/or methods to advance restart or debug state.
Huawei's approach appears to be consistent with this approach as it only uses a sentinal value of zero as significant in the field.
It should be made explicit that directly manipulating the field to a given value does not guarantee resumption from what might appear to be the uOP number in that value.

3) I fully agree that the three approaches proposed are an excellent starting point for evaluating a restart and debug mechanism.

## There are also a few typos in 
Huawei Custom Extension/riscv_uop_handling.rst

### Move explanation into parentheses, change:
>It is possible to implement instructions which require more than two register read ports as a sequence of UOPs, 
providing any intermediate results can be written to standard registers (not special temporary registers) as this
would prevent the UOP using a standard RISCV encoding.

to
>It is possible to implement instructions which require more than two register read ports as a sequence of UOPs, 
providing any intermediate results can be written to standard registers (not special temporary registers as this
would prevent the UOP using a standard RISCV encoding).

### negative offset for   
```
UOP 2 lw  x18, -26(sp);
```
should be
```
 UOP 2 lw  x18, -16(sp);
```

tariqkurd-repo, 2020-09-15:
Hi David,

Thanks for very useful comments. The UOP restart mechanism is a report of what we did to get it working in a simple, I don't expect it to be the actual RISC-V solution, there was no way I'd get away with allocating bits in mstatus.
I've fixed the issues you pointed out - but which typos where you referring to?

Tariq

tariqkurd-repo, 2021-01-19:
Hi @David-Horner can I close this one? The current push/pop proposal doesn't make any use of the document referred to in this issue.

closed 2021-01-19 by David-Horner