riscvarchive/riscv-code-size-reduction #3: Create a timebox tag

cetola, 2020-09-14:
In repos like this one where conversations often take place in the issues tab, I recommend folks use a "timebox" tag and decide on the timeout period.

This keeps issues from remaining open for long periods of time with no activity and no answers. I'd recommend the timeout period to be 1 week. After that, if no one has commented or answered the question, it is best to move that question to the mailing lists.

cetola, 2020-09-14:
@tariqkurd-repo the repo is yours so I'll let you decide. :)

tariqkurd-repo, 2020-09-15:
Hi Stephano,

That sounds good to me, how do I do it?

Tariq

cetola, 2020-09-15:
Created and applied to https://github.com/riscv/riscv-code-size-reduction/issues/1

If there is no more discussion after a week or so I'll move that question to the public mailing list:
https://groups.google.com/a/groups.riscv.org/forum/#!forumsearch/

Probably HW-DEV

closed 2020-09-15 by cetola