riscvarchive/riscv-code-size-reduction #9: Idea: Add multiply-and-accumulate operations extension ("Zmac")

stnolting, 2020-11-01:
Even pretty small architectures (like the 16-bit msp430) provide multiply-and-accumulate (MAC) operations,
but RISC-V lacks this feature. I think MAC operations are very helpful for general purpose computing - not only for DSP applications.

How about something like a `"Zmac"` extension - `M` extension with **ac**cumulator support?

In order to show the `Zmac` performance increase - at least from my point of view :wink: - here is a simple example that computes the scalar product of two vectors `vector_a` and `vector_b`:

```C
  int i;
  volatile int32_t vector_a[64];
  volatile int32_t vector_b[64];
  int64_t accu = 0;
  
  for (i=0; i<64; i++) {
    accu += ( (int64_t)vector_a[i] ) * ( (int64_t)vector_b[i] );
  }
```

For a `rv32im` architecture this translates into:

```
  164:  df010113    addi  sp, sp, -528
  168:  20112623    sw    ra, 524(sp)
  16c:  00000513    li    a0, 0            // initialize accu.LOW to zero
  170:  00000593    li    a1, 0            // initialize accu.HIGH to zero
  174:  00000793    li    a5, 0
  178:  04000813    li    a6, 64
mac_loop:           
  17c:  00279713    slli  a4, a5, 0x2
  180:  20010693    addi  a3, sp, 512
  184:  00e68733    add   a4, a3, a4
  188:  e0072683    lw    a3, -512(a4)     // get vector_a[i]
  18c:  f0072603    lw    a2, -256(a4)     // get vector_b[i]
  
  190:  00178793    addi  a5, a5, 1        // increment i
  
  194:  02c68733    mul   a4, a3, a2       // compute mul_res.LOW
  198:  02c696b3    mulh  a3, a3, a2       // compute mul_res.HIGH
  
  19c:  00e50733    add   a4, a0, a4       // accumulation begin
  1a0:  00a73633    sltu  a2, a4, a0       // ...
  1a4:  00070513    mv    a0, a4           // ...
  1a8:  00d585b3    add   a1, a1, a3       // ...
  1ac:  00b605b3    add   a1, a2, a1       // accumulation end
  
  1b0:  fd0796e3    bne   a5, a6, 17c <mac_loop>
```

Here, the accu is mapped to general purpose registers `a1:a0` and each accumulation takes five instructions.

If we add an accumulator and an adder to the multiplier unit, the `M` extension could also perform MAC operations.
For instance, the MAC-capable `mul` instruction would do something like this (the accumulator could be cleared if `rs1` and `rs2` are both zero):

```
  uint_xlen_t mul(uint_xlen_t rs1, uint_xlen_t rs2, bool is_mac)
  {
    if ((rs1 == zero) && (rs2 == zero) && (is_mac)) {  // clear accu?
      accu = 0;  // accu has a size of 2 * uint_xlen_t
    }
 
    if (is_mac) {
      accu = accu + ((2*uint_xlen_t)rs1 * (2*uint_xlen_t)rs2);
    }
    else {
      accu = 0    + ((2*uint_xlen_t)rs1 * (2*uint_xlen_t)rs2);
    }
 
    return (uint_xlen_t)(accu & 0xFFFFFFFF)
  }
```

If `is_mac` is `false` all multiplication operations act like "normal" multiplication operations.

One could add a `.mac` suffix to the standard multiplication operations (`mul` `mulh` `mulhsu` `mulhu`) to enable the
accumulator (setting `is_mac` = `true`). All multiplication operations us `funct7 = 0000001` and I think there is enough opcode space left to encode the `.mac` option.

The resulting program (`rv32im_Zmac`) could be something like this:

```
  178:  ????????    mul.acc zero, zero, zero   // clear accumulator
mac_loop:
  17c:  00279713    slli  a4, a5, 0x2
  180:  20010693    addi  a3, sp, 512
  184:  00e68733    add   a4, a3, a4
  188:  e0072683    lw    a3, -512(a4)         // get vector_a[i]
  18c:  f0072603    lw    a2, -256(a4)         // get vector_b[i]
  
  190:  00178793    addi  a5, a5, 1            // increment i
  
  194:  ????????    mul.acc   a0, a3, a2       // compute multiplication and update internal accu, get low part of accu into a0
  
  198:  --------    bne   a5, a6, 17c <mac_loop>
mac_loop_end:
  19c:  ????????    mulh.acc  a1, a3, zero // get high part of accu
```

Only **one** instruction (`mul.acc`) for the entire MAC operation plus a `mulh.acc` after the loop to get the high part of the accu.
Also, this approach reduces register pressure as we need less temporary registers.

I know that there are custom extension for MAC operations like in the PULP RI5CY processor. But their approach
requires an additional register file read-port (they do `rd = rd + rs1 * rs2`). In contrast to `Zmac*, this increases register file complexity and also does not provide a 64-bit accu option on 32-bit platforms.

Maybe I am overlooking something and the `Zmac` concept is better off as a custom extension. Or the MAC operations
are already a part of another DSP-like extension draft... I don't know :thinking:

What do you think?

jnk0le, 2020-11-01:
There is a P extension that covers this field. 
https://github.com/riscv/riscv-p-spec
You shuld try there if your approach differs from current spec.

stnolting, 2020-11-01:
Thanks @jnk0le, I wasn't aware of `P` extension's MAC instruction.

However, I see a problem with the `P` extension here. First of all this extensions is pretty heavy. I am not sure if there are any subsets of it (yet), but especially for small and deeply-embedded processors `P` seems to be pretty huge.

Also, instructions like `KMMAC rt, ra, rb` are using `rt` also as source operand (as far as I can see...), which requires additional register file read ports.

Furthermore, I cannot find a suitable non-SIMD instruction that allows to compute `A[63:0] = A[63:0] + B[31:0] * C[31:0]` on `rv32` platforms and `A[127:0] = A[127:0] + B[63:0] * C[63:0]` on `rv64` platforms.

When it comes to a silicon area vs. performance trade-off `Zmac` seems to be more reasonable for small-scale cores (just my opinion). 🤔

Rolfy360, 2020-11-01:
This kind of instructions are similar to MADD/MSUB in MIPS or SMULL/UMULL in ARM.
For lightweight usage, integer multiply-add should be put in M-extension.

stnolting, 2020-11-03:
So, in summary I would like to suggest four MAC instructions for the Zmac extension:

```c
uint_xlen_t mul.acc(uint_xlen_t rs1, uint_xlen_t rs2)
{
  if (rs1 == zero) {
    accu[uint_xlen_t-1:0] = rs2; // initialize accu.low
  }
  accu = accu + rs1 * rs2;
  return (uint_xlen_t)(accu[uint_xlen_t-1:0]); // return accu.low
}
```

```c
uint_xlen_t mulh.acc(uint_xlen_t rs1, uint_xlen_t rs2)
{
  if (rs1 == zero) {
    accu[2*uint_xlen_t-1:uint_xlen_t] = rs2; // initialize accu.high
  }
  accu = accu + rs1 * rs2;
  return (uint_xlen_t)(accu[2*uint_xlen_t-1:uint_xlen_t]); // return accu.high
}
```

And accordingly for `mulhu.acc` and `mulhsu.acc`.

In this context `accu` is a "global variable" (a dedicated hardware register in the multiplier unit, respectively) with a size of `2*uint_xlen_t`.

Besides the actual MAC operations, the following operand combinations could be used to directly interact with the accumulator:
* `mul.acc  zero, zero, rs` - initialize accu.low with `rs` (`rs = zero` can be used to clear accu.low)
* `mulh.acc zero, zero, rs` - initialize accu.high with `rs` (`rs = zero` can be used to clear accu.high)
* `mul.acc  rd, zero, rs` - copy accu.low to `rd`, `rs` can be any register but not `zero`; does not modify accu
* `mulh.acc rd, zero, rs` - copy accu.high to `rd`, `rs` can be any register but not `zero`; does not modify accu

Any thoughts? ;)

jnk0le, 2020-11-03:
> Furthermore, I cannot find a suitable non-SIMD instruction that allows to compute `A[63:0] = A[63:0] + B[31:0] * C[31:0]` on `rv32` platforms

Prior presentations mentioned exactly this kind of operation https://youtu.be/tMhnvfbPM3A?t=120.
Maybe it was dropped later for some reason.

> When it comes to a silicon area vs. performance trade-off `Zmac` seems to be more reasonable for small-scale cores (just my opinion). 🤔

OTOH, non-reg accumulator adds extra state to save/restore in interrupts.

tariqkurd-repo, 2020-11-12:
Hi @stnolting,

This isn't the right place for a request like this. Zmac will only have a small effect on code size. A tiny proportion of real executable code will contain multiply-accumulates.
If you want to request this functionality I suggest you go email the request to either the P-extension to add functionality or a subset, or the unprivleged committee to add funcitonality to the M-extension.

If it's ok with you I'll close this issue.

stnolting, 2020-11-12:
> If it's ok with you I'll close this issue.

Sure. Thanks for hint! :wink:

closed 2020-11-12 by stnolting

jnk0le, 2021-01-04:
> Prior presentations mentioned exactly this kind of operation https://youtu.be/tMhnvfbPM3A?t=120.
> Maybe it was dropped later for some reason.
it was not dropped, there is SMAR64 and UMAR64 that does exactly this.