riscvarchive/riscv-code-size-reduction #10: Illegal Instruction and Table 3 Definition

yichanfeng, 2020-11-02:
Hi,

I have two questions about the "illegal instruction" and "Table 3" definition.

Q1. 
In "Table 3", it said that rcount 8-10 are EABI only. 
But, "RV32I EABI/RV64I EABI with rcount 8-10" are marked as "N/A" in "Table 3". 
=> Is it a typo ?

Q2.
In section 3.8.2, it said that RV32E and RV64E take an illegal instruction exception if rcount > 1 && rcount < 8.
But, "RV32E/RV64E with rount **2**" is not marked as "reserved". 
=> Do we need to change the illegal instruction condition to "rcount > **2** && rcount < 8" ?

Thanks!

tariqkurd-repo, 2020-11-12:
sorry for the late replay - yes you're right! I've fixed table 3, can you check it please?
辛苦你了

yichanfeng, 2020-11-13:
Hi, 
Thanks for the update on the issue.

a. Modification of Table 3 is correct for Q1.
b. For Q2, the description of section 3.8.2 still does not match the definition of Table 3. 
    Could you please check it again ? Thanks!

tariqkurd-repo, 2020-11-13:
You're right! I've fixed it. If it's ok please close the issue.

Tariq

yichanfeng, 2020-11-16:
Hi,
Thanks for the update on the issue.
Modification of section 3.8.2 is correct for Q2.

closed 2020-11-16 by yichanfeng