riscvarchive/riscv-code-size-reduction #11: RV64: L.LUI?

tovine, 2020-11-05:
In your 48-bit encodings proposal the L.LI instruction that loads a sign-extended 32-bit immediate into bits [31:0] of `rd`; have you given any thought to a corresponding L.LUI  instruction for RV64, that loads the 32-bit immediate into bits [63:32]?

Seems like a natural complement to this instruction, though I'm not sure how much it would save in terms of code size or if it's even relevant (do machines typically use 64-bit addressing?). Anyway - it's just a thought, didn't get time to throw it out there in the meeting on Tuesday.

tovine, 2020-11-05:
Alternatively, if 64-bit memory addressing isn't a thing, maybe loading the 32-bit immediate into bits [47:15] or [44:12] or something, in order to directly address a larger range with load, store and branch instructions.

tariqkurd-repo, 2020-11-12:
Hi - interesting request.
I think you can get the same functionality by using a 16-bit shift instruction after the L.LI instruction, so maybe it's not worth it.
I'll bear it in mind when we come to benchmark code compiled for RV64,w hich we haven't done yet.
Is it OK to close the issue?

tovine, 2020-11-12:
Right, that makes sense (also a bit more flexible than my proposal).
It would probably only cover a few corner cases, and then only save 2 bytes (for the `C.SLLI`).

Closing it for now is fine by me, and revisiting it when we start tackling RV64 sounds reasonable. 🙂

closed 2020-11-12 by tovine