riscvarchive/riscv-code-size-reduction #13: Can we drop POP in favor of POPRET to save 16-bit encoding space?

chuanhua, 2020-11-17:
If we have POPRET, there is a concern that the POP instruction will not be used most of the time.
What is the code size saving percentage between POPRET and POP?
Can we remove POP and just have POPRET to save 16-bit encoding space?

Thanks.

tariqkurd-repo, 2020-11-18:
Interesting point...... I had a look at the IoT code, there are lots of sequences like this:

0100f8a8 <function>:
 100f8a8:	8018                	push	{ra},-16
...
 100f900:	8010                	pop	{ra},16
 100f902:	5742effb          	j16	e3de76 <function2>

so pop is used, and the ret comes from function2, not from function.

there are ~ 6000 POPRETs and ~ 2200 POPs, and it's true that we can probably encode the most useful POPs in different unused rcount values of POPRET, and save encoding space overall.
I'll add this as an issue and look at this once we have more benchmarks running.

tariqkurd-repo, 2020-12-07:
Hi Chuanhua,

I've reduced the encoding space takenup by POP to the most frequent ones only.
If that's ok can you close the issue?

Tariq

David-Horner, 2020-12-07:
I am wary of encoding decisions based on limited assessments.
We cannot prognosticate what the eco system will have accomplished by next year, or 2 or 5 years off.
Unless there is a compelling reason for these choices over just current frequency of use, I advise the issue remain open until we have validated that these proposed encodings are indeed optimal for some agreed upon criteria.

tariqkurd-repo, 2020-12-07:
Hi David, I agree. What I propose is those extra encodings remain reserved so we can add them later (I'll add this text into the doc). I.e. we can add full reg lists into C.POP, or wider SPIMM values in the future once we have a better idea of the actual usage in the real world.

tariqkurd-repo, 2020-12-09:
as it stands, allocating another reserved bit has minimal impact on the code size, it certainly doesn't double the benefit. IN the future though, who knows so it's best to leave them reserved for the time being. I'll close this issue if that's ok.

closed 2020-12-09 by tariqkurd-repo