riscvarchive/riscv-code-size-reduction #15: Clarifications needed

allenjbaum, 2020-12-01:
In section 2, for 26bit encodings
    "fs0-fsM may not be included"
should be
  "fs0-fsM must not be included"

 "the stack_adjustment must be no more than the total memory required for xreg_list, 
   rounded up to a multiple of 16-bytes, + 7*16-bytes"

This is explained later, but it should be clarified heere:

 "the stack_adjustment must be no more than the total memory required for xreg_list, 
   rounded up to a multiple of 16-bytes, plus an additional (0.. 7)*16-bytes"
Ditto for the 32B version, except (0..15)
------
In sec 3.1, push pseudocode, you decrement the stack pointer immediately.
If you then trap in the middle, you need to undo that before returning. 
Otherwise, you can simply return, and re-do the entire operation.

Also, you have code to detect misaligned loads/store or fp doubles by adjusting the address.
IS this a security hole - should there be a store operation to clear out the old contents?.
Is this an issue for the additional frame space being allocated - should we be worrying about this?
------
in sec. 4, you have separate encodings for .F vs .D. IS that necessary?
If the D-extension is present, you could always save and restore the full 8 bytes and save the opcode space. Keeping track of which registers contain single vs. double doesn't sound easy, and there will be some performance lost if  the formats are mixed - that's the downside.

David-Horner, 2020-12-02:
@allenjbaum IS this a security hole - should there be a store operation to clear out the old contents?.
Is this an issue for the additional frame space being allocated - should we be worrying about this?

There is no security hole to allocate unused stack space.
The security hole is: not sanitizing sensitive data in memory after it is no longer needed.
The O/S should only allocated cleansed data for stack use, or for any use in a secured environment.
These instructions need not and should not be burdened with the task.

However, a variant of this instruction that zero fills the space may be more performant.
If a whole cache line is written then the remote data need not be read into the local cache.

**I am in favour that such instructions, that advance the sp, are agnostic with regards to "gaps".**
The implementation may write "ones" [binary zero] or leave undisturbed "unwritten" allocations.
A given implementation can write some allocated with zeros and leave others undisturbed.
In general, the hardware is aware of the cache line state, it can complete the cache-line filling the gaps with ones to optimize bus traffic, or if the cache-line is resident avoid the write and save write-buffer space [or anything in between, such as flagging bytes to be ignored by the cache recall]. As the stack allocation can spen multiple cach-line that strategy may be different for each cache-line affected [some gap fill, some not write to a gap, some flag as agnostic and the cache controller itself cache-line fill with  some "poison" value like all ones.

See this vector discussion for agnostic suggestion:
https://github.com/riscv/riscv-v-spec/issues/367#issuecomment-594786215
The whole discussion is informative and relevant to managing "gaps".
https://github.com/riscv/riscv-v-spec/issues/367

Note: I anticipate no application usefulness for retaining gaps of newly allocated stack space.
Indeed some O/S may aggressively recover stack allocations [a bad thing without application participation] but certainly from a page out perspective, which also makes the zeroing potentially less performant if the allocated space in a "new" page is speculatively allocated and not actually needed by the application. The hardware could optimize this situation and avoid retrieval from disk swap space [details by request]

allenjbaum, 2020-12-02:
> riscv/riscv-v-spec#367 (comment)
> riscv/riscv-v-spec#367 <https://github.com/riscv/riscv-v-spec/issues/367>

OK, so this is basically application managed security, much like
application managed coherence - if you care, you'll do the right thing. I'm
fine with that.
From an ABI point of view, you can initialize gaps or not, and the
application shouldn't care.
A security oriented core might choose to initialize it if it wants to.

David-Horner, 2020-12-02:
On 2020-12-03 2:18 a.m., Allen Baum wrote:
> A security oriented core might choose to initialize it if it wants to.

I agree.

There may be performance loss, and potential page fault as filling is no 
longer speculative,
but agnostic "gap" handling allows either.

tariqkurd-repo, 2020-12-07:
Looks like I wasn't "watching" this repo correctly so I didn't see this thread, sorry!
At the moment the compiler allocates stack space, stores some registers and zeroes a few locations where necessary.
My proposal is to add PUSHZERO which will either store or zero all the stack space, to avoid the zero intialisation of variables, which wastes code-size. PUSHZERO would store zero to the gap.

For PUSH.F/PUSH.D - they are necessary, otherwise we can't decode the instruction without checking MISA.F/D to know which extensions are enabled/implemented as the behaviour would then vary dynamically. It's also useful because you might be running an SP algorithm on a RV32D core and not want to save/restore the upper halves of the registers.

>If the D-extension is present, you could always save and restore the full 8 bytes and save the opcode space. Keeping track of >which registers contain single vs. double doesn't sound easy, and there will be some performance lost if the formats are mixed - >that's the downside.

As stated about the D-extension can be dynamically disabled, which is the problem.

I've fixed your other comments Allen, thanks.

tariqkurd-repo, 2020-12-11:
For PUSH.F/PUSH.D the stack adjustment is different depending on the state of MISA.D, and so it is not possible to *decode* the instruction (as the syntax includes the stack adjustment) without knowing the state of MISA.D, and by extension MISA.Q. Therefore we need the separate opcode to make the encoding unambiguous only because MISA.D / Q are writable. We can't just save/restore the full width because that extension may be disabled. Hope that's ok - closing it now, I'll take up clarifying the policy on controlling instruction behaviour by using CSRs with Ken Dockser.

tariqkurd-repo, 2020-12-11:
You're right - in theory MISA.MXL could change and then that would also change the decoding as the stack size would change.
However I wouldn't expect any change in MISA.MXL, whereas MISA.D=0 could be used for lazy context save/restore, but if that's happening then it wouldn't save any F registers at all, as oppose to the least significant half of them.

If it's useful to save/restore F or D then we either need an instruction variant, or make it automatic based upon MISA.D. If we make it automatic then the decode should always assume that MISA.D is enabled, which would certainly save encoding space.

Is that reasonable?

allenjbaum, 2020-12-13:
Why would you set MISA.D=0 rather than fast stud.fs=off or clean or initial?

-Allen

David-Horner, 2020-12-13:
Mstatus and sstatus field FS. ( and H-extension *status FS as applicable  )
Not sure if stud is a pseudo register?

On Sat, Dec 12, 2020, 23:49 Allen Baum, <notifications@github.com> wrote:

> Why would you set MISA.D=0 rather than fast stud.fs=off or clean or
> initial?
>
> -Allen
>
>
>

allenjbaum, 2020-12-13:
Sorry, auto correct. Just happened again: doesn’t like “fstatus”. I forget if it’s in fstatus.fs, but is in mstatus 

-Allen

tariqkurd-repo, 2021-01-08:
I think I can close this one now as I have PUSH.[FDQ] opcodes.
PLease let me know if I need to reopen it.

Tariq

closed 2021-01-08 by tariqkurd-repo