riscvarchive/riscv-code-size-reduction #17: issues need to be clarified

yichanfeng, 2021-01-07:
Hi,

I have few questions about the ISA spec [(riscv_push_pop_extension_RV32_RV64_UABI.adoc).](https://github.com/riscv/riscv-code-size-reduction/blob/master/ISA%20proposals/Huawei/riscv_push_pop_extension_RV32_RV64_UABI.adoc)

Q1.
According to the description in chapter 2 **"Assembler Syntax"**, the f register should be listed after x register or ra.
`<fxreg_list> ::= <ra> "," <xreg_range> ", " <freg_range>`

But, it seems that the example code in section 9.4 **"PUSH.F RV32 UABI (16-byte aligned stack)"** does not match the above description.
`push.f  {fs0, ra, s0-s4}, -64`  => **fs0**(f register) is listed **before** x register or ra => Is it a legal syntax ?

Q2.
In section 4.1 **"PUSH Pseudo-code"** and section 4.2 **"POP[RET] Pseudo-code"**, it said that 

> if FLEN > XLEN, then get the address back into alignment before saving the F registers.

But, it seems that the proposed pseudo-code cannot guarantee that the memory address for each f register access is always **aligned to FLEN** if the alignment-while-loop is **outside** of the for-loop. 

Please see the following example =>

> [RV32D, XLEN=32, FLEN=64]: 
bytes = 4, 
addr = 0x104(before while loop) -> 
addr = 0x100(after while loop) -> 
addr = 0xfc (the 1st f register store, **not** FLEN aligned) -> 
addr = 0xf8 (the 2nd f register store, FLEN aligned) -> 
addr = 0xf4 (the 3rd f register store, **not** FLEN aligned) -> 
...

> if` (freq_list) {
  switch (opcode) {
    //if FLEN > XLEN, then get the address back into alignment before saving the F registers
    "PUSH.D": while(addr% 8) {addr-=bytes;}
    "PUSH.Q": while(addr%16) {addr-=bytes;}
  }
  for(i in freg_list)  {
    addr-=bytes;
    switch (opcode) {
      "PUSH.F": asm("fsw fs[i], 0(addr)");
      "PUSH.D": asm("fsd fs[i], 0(addr)");
      "PUSH.Q": asm("fsq fs[i], 0(addr)");
    }
  }

Q3.
In addition to Q2's alignment issue, do we need to adjust the equation for **stack_adjustment** calculation if the required stack space of one f register is not equal to XLEN (i.e., FLEN != XLEN) ?

> total_register_bytes = number_of_registers_in_reg_list * XLEN / 8

Q4.
In section 9.4 **"PUSH.F RV32 UABI (16-byte aligned stack)"**, it seems that the **address offset** described in example code does not match the pseudo code.
//Current
> push.f  {fs0, ra, s0-s4}, -64
addi sp, sp, -64;
sw  ra,  28(sp);  //**not** equal to "sp - 4"
sw  s0,  24(sp); //**not** equal to "sp - 8"
sw  s1,  20(sp); //**not** equal to "sp - 12"
sw  s2,  16(sp); //**not** equal to "sp - 16"
sw  s3,  12(sp); //**not** equal to "sp - 20"
sw  s4,   8(sp); //**not** equal to "sp - 24"
fsw fs0,  4(sp); //**not** equal to "sp - 28"

//Modified
> push.f  {fs0, ra, s0-s4}, -64
addi sp, sp, -64;
sw  ra,  60(sp);  //equal to "sp - 4"
sw  s0,  56(sp); //equal to "sp - 8"
sw  s1,  52(sp); //equal to "sp - 12"
sw  s2,  48(sp); //equal to "sp - 16"
sw  s3,  44(sp); //equal to "sp - 20"
sw  s4,   40sp); //equal to "sp - 24"
fsw fs0,  36(sp); //equal to "sp - 28"

Q5.
In ISA SPEC [riscv_push_pop_extension_RV32_RV64.adoc](https://github.com/riscv/riscv-code-size-reduction/blob/master/ISA%20proposals/Huawei/riscv_push_pop_extension_RV32_RV64.adoc), section 3.8.1 **"Stack pointer misalignment"** said that 

> Stack pointer misalignment causes a Store Access Fault for C.PUSH[.E] or a Load Access Fault for C.POP[.E]/C.POPRET[.E].
Take an exception if:
RV64: The stack pointer is not 16-byte aligned
RV32: The stack pointer is not 8-byte aligned and rcount < 2 || rcount > 7, otherwise if the stack pointer is not 16-byte aligned

Do we need to add the similar **alignment check**  for the UABI-only ISA SPEC (riscv_push_pop_extension_RV32_RV64_UABI.adoc) ?  

Thanks!

yichanfeng, 2021-01-07:
Hi 

I have one suggestion about "assembler syntax" .

For the test pattern development or writing assembly code, it is not so straightforward to calculate the **legal space range** for **"stack_adjustment".**

Perhaps it would be better if we could change the **"assembler syntax"** from 
`push/pop {<xreg_list> | <fxreg_list>}, +/-<stack_adjustment>` to 
`push/pop {<xreg_list> | <fxreg_list>}, +/-<spimm>`.

Please see the following example.

//Current: calculate stack_adjustment (64 )
`c.push  {ra, s0-s5}, -64
Encoding: rcount=5, spimm=2
`

//Suggested: only assign spimm value for the additional stack space (2)
`c.push  {ra, s0-s5}, -2 
Encoding: rcount=5, spimm=2
`
Thanks!

tariqkurd-repo, 2021-01-08:
Hi,

>push.f {fs0, ra, s0-s4}, -64 => fs0(f register) is listed before x register or ra => Is it a legal syntax ?

you're right - I've fixed the examples

For the alignment for the F registers - again you're right - the stack decrement was wrong - I've fixes the psuedo-code

For the stack pointer alignment check - the code we are replacing with PUSH/POP doesn't check alignment of the stack pointer, but if the core doesn't support misaligned load/stores then they will generate the alignment faults, so there doesn't seem to be much need to check the SP alignment.

For the stack adjustment value in the assembly syntax: I agree it's a bit tricky to determine the legal values of SPIMM. When the specification is frozen I can add a lookup table showing the legal values for different numbers of registers in the list- but we feel that showing the total stack adjustment is the best approach for the compiler engineer and the assembly programmer.

Please check my changes to the document, and thanks for the valuable input

Tariq

yichanfeng, 2021-01-11:
Hi Tariq,

Thank you for the quick reply.

Please see the following results after carefully checking the new changes.
Q1: **OK**

Q2: **OK**

Q3: Because the pseudo code of F register access has been changed, it seems that we need to modify the following equation to calculate the correct stack size (consider XLEN and FLEN instead of XLEN only) ?

> total_register_bytes = number_of_registers_in_reg_list * XLEN / 8

Q4:  The address offset described in example code does not match the pseudo code.
        => the first effective address for push is not equal to **sp-4**

Q5: **OK**

BTW, there is another TYPO in example code **"9.1. C.PUSH RV32 UABI (16-byte aligned stack)"**.

I think the correct value of spimm should be **1.**
> Encoding: rcount=5, spimm=2

Thanks!
Yi-Chan.

tariqkurd-repo, 2021-01-11:
Hi,

**total_register_bytes = number_of_registers_in_reg_list * XLEN / 8**
-> fixed

**=> the first effective address for push is not equal to sp-4**
-> fixed

**I think the correct value of spimm should be 1.**
I think this one is correct:

c.push  {ra, s0-s5}, -64
Encoding: rcount=5, spimm=2

7 registers, 7*4=28, this is rounded up to 32
the total stack adjustment requires 2x16 to get an extra 32, so spimm=2
is that ok?

yichanfeng, 2021-01-11:
Hi Tariq,

Thank you for the quick reply.

Yes, you are right for the example code "9.1. C.PUSH RV32 UABI (16-byte aligned stack)". 

It’s my mistake. Actually, I would like to discuss example code **"9.2. C.PUSH RV32I UABI (16-byte aligned stack)"**.

=> 3 registers, 3*4=12,  this is rounded up to 16. 
=> the total stack adjustment requires 2x32 to get an extra 16, so spimm should be **1**

> c.push {ra, s0-s1}, -32
Encoding: rcount=2, spimm=2

tariqkurd-repo, 2021-01-11:
ok, fixed! is it ok now?

yichanfeng, 2021-01-11:
Hi Tariq,

It's OK. Many thanks! 

I will close this issue.

Yi-Chan.

yichanfeng, 2021-01-11:
Hi Tariq,

Sorry, I would like to reopen this ticket for the issue of new pseudo code after discussion with my colleague.
1) assume that the HW configuration is RV32D (i.e XLEN = 32, FLEN=64)
2) assume that the variable **addr** is 0x10c before executing the following pseudo code of F register
3) execute while loop to get the 8-byte aligned address => 0x10c (before while loop) -> 0x108 (after while loop)
4) Then, processor executes the 1st micro-op "PUSH.D" to store a 64-bit F register to memory address 0x108.
     => The problem is that the above micro-op will **overwrite** the 32-bit X register data stored in memory 0x10C.
     => It seems that the pseudo code of address decrement (addr-=4/8/16) should be executed before FP store instruction to 
           prevent overwriting stack.

> if (freq_list) {
  switch (opcode) {
    //if FLEN > XLEN, then get the address back into alignment before saving the F registers
    "PUSH.D": while(addr% 8) {addr-=bytes;}
    "PUSH.Q": while(addr%16) {addr-=bytes;}
  }
  for(i in freg_list)  {
    switch (opcode) {
      "PUSH.F": {asm("fsw fs[i], 0(addr)");addr-=4;}
      "PUSH.D": {asm("fsd fs[i], 0(addr)");addr-=8;}
      "PUSH.Q": {asm("fsq fs[i], 0(addr)");addr-=16;}
    }
  }

Thanks!
Yi-Chan.

tariqkurd-repo, 2021-01-11:
thanks! you're right. I've fixed it - can you check?

yichanfeng, 2021-01-12:
Hi Tariq,

Thank you for the quick reply.

After checking the latest fixes, there are three issues.

1) Only **"4.2. POP[RET] Pseudo-code"** has been fixed. 
    => I think the fixes should be applied to **both sides** (PUSH + POP).

2) In **"4.2. POP[RET] Pseudo-code"**, the following **switch case** has the wrong name of micro op.
    => **"PUSH.F/D/Q"** should be replaced with **"POP.F/D/Q"**

> 4.2. POP[RET] Pseudo-code
for(i in freg_list)  {
    switch (opcode) {
      "PUSH.F": {addr-=4;asm("flw fs[i], 0(addr)");}
      "PUSH.D": {addr-=8;asm("fld fs[i], 0(addr)");}
      "PUSH.Q": {addr-=16;asm("flq fs[i], 0(addr)");}
    }
  }

3) There is another issue caused by **while-loop** of F-register alignment. 
    => The following equation **cannot** calculate the **correct stack space** used for registers under some special cases.
    > total_register_bytes = number_of_registers_in_reg_list * XLEN/8 + number_of_registers_in_freg_list * FLEN/8
       stack_adjustment = Align16(total_register_bytes) + 16*spimm

    => Example is described below
    a) RV32D (i.e., XLEN=32, FLEN=64) & push 3 register (ra, s0, fs0) to stack
    b) According to above equation, the **Align16(total_register_bytes)** = Align16(2*4 + 1*8) = 16 bytes.
        However, the actual stack space used for X and F register is **20 bytes (16 + 4)**.
        => _the extra 4-byte for alignment is not counted._

    --------------  0x114 (SP)
        ra (XLEN)
    --------------  0x110 (SP-4)
        s0 (XLEN)
    --------------  0x10C (SP-8, not 8-byte aligned)
        alignment
        (while-loop)
    --------------  0x108 (SP-12, 8-byte aligned)
       fs0 (FLEN)
    --------------  0x100 (SP-8)

Howard

tariqkurd-repo, 2021-01-12:
You're right again, I've fixed it again. Please check.

yichanfeng, 2021-01-13:
Hi Tariq,

Thank you for the quick reply.

After checking the latest fixes, the revised pseudo code is OK. 

However, the "padding" issue for stack adjustment still remains.

> if (FLEN>XLEN && number_of_registers_in_freg_list>0) {
  //need padding to ensure FLEN registers are aligned
  padding = (FLEN/XLEN - (number_of_registers_in_reg_list % (FLEN/XLEN))) * XLEN/8;
} else {
  padding = 0
}

////////////////////////////////////////////////////////////
**Case 1. FLEN (64) > XLEN(32) & initial value of SP is FLEN aligned**
=> In fact, "padding" should be **ZERO**. But, the calculated "padding" is **8** bytes. 
=> (64/32 - 2 %(64/32)) * 32/8 = (2-0) * 4 = 8

-------------- 0x110 (SP)
ra (XLEN)
-------------- 0x10C (SP-4)
s0 (XLEN)
-------------- 0x108 (SP-8, 8-byte aligned) => while-loop is **not** required
fs0 (FLEN)
-------------- 0x100 (SP-16)
////////////////////////////////////////////////////////////
**Case 2. FLEN (64) > XLEN(32) & initial value of SP is not FLEN aligned**

I think that the formula should consider **the initial value of SP**. 
Otherwise, it cannot correctly calculate the padding value for the following example.
=> In fact, "padding" should be **4** byte. But, the calculated "padding" is **8** bytes. 
=> (64/32 - 2 %(64/32)) * 32/8 = (2-0) * 4 = 8

-------------- 0x114 (SP)
ra (XLEN)
-------------- 0x110 (SP-4)
s0 (XLEN)
-------------- 0x10C (SP-8, not 8-byte aligned)
alignment
(while-loop)
-------------- 0x108 (SP-12, 8-byte aligned)
fs0 (FLEN)
-------------- 0x100 (SP-20)

////////////////////////////////////////////////////////////
If possible, a **dedicated** instruction for F-register stacking is better than merging them (X+F REG) together to a single push/pop instruction.
=> A single push/pop instruction will make the HW implementation become overly complicated, especially the padding value calculation.

Thanks!
Yi-Chan

yichanfeng, 2021-01-13:
Hi Tariq,

I also found a new TYPO in **"9.5. PUSH.D RV64 UABI (16-byte aligned stack)"**.
=> The following micro op should be changed from "sw" to "sd" for RV64 configuration.

> push.f  {ra, s0-s4, fs0}, -64
addi sp, sp, -64;
sw  ra,  56(sp);
sw  s0,  48(sp);
sw  s1,  40(sp);
sw  s2,  32(sp);
sw  s3,  24(sp);
sw  s4,  16(sp);
fsd fs0,  8(sp)

Thanks!
Yi-Chan

tariqkurd-repo, 2021-01-13:
Hi - I've fixed the calculation.
In fact the calculation is very simple for RV32D and RV64Q, only RV32Q is mroe complex.
Please see the examples in the text.
The objective is to save code-size so one instruction will save X and F registers.

Thanks for your input!

yichanfeng, 2021-01-14:
Hi Tariq,

Thank you for the quick reply.

After checking the latest fixes, the "padding" issue for stack adjustment still remains if the initial value of SP is not 16-byte aligned or FLEN aligned.

=> Please see the following example, it will violate the rule defined **"Table 3. RV32D padding examples, XLEN=32, FLEN=64"**.

> #Xreg: 2n    -> padding = none
#Xreg: 2n+1   -> padding = 4-byte

------------------------------------------------------------------
[RV32D]: FLEN (64) > XLEN(32) & initial value of SP is not FLEN aligned

I think that the formula should consider the initial value of SP.
Otherwise, it cannot correctly calculate the padding value.
=> In fact, "padding" should be 4 byte. But, the "padding" value from "Table 3" is 0 bytes 
                                                                (#Xreg = {ra, s0} = 2 -> padding(2n) = none).

-------------- 0x114 (SP)
ra (XLEN)
-------------- 0x110 (SP-4)
s0 (XLEN)
-------------- 0x10C (SP-8, not 8-byte aligned)
alignment
(while-loop)
-------------- 0x108 (SP-12, 8-byte aligned)
fs0 (FLEN)
-------------- 0x100 (SP-20)

Yi-Chan

tariqkurd-repo, 2021-01-14:
Hi - you're right. An earlier version of the spec had an illegla instruction exception  if SP wasn't 16-byte aligned. I removed it, but now it has a purpose because the padding calculation needs it. Please check it's ok.

yichanfeng, 2021-01-15:
Hi Tariq,

The new constraint (16-byte aligned) is OK. Many thanks!

BTW, I also found a new TYPO in "9.5. PUSH.D RV64 UABI (16-byte aligned stack)".
=> The following micro op should be changed from "sw" to "sd" for RV64 configuration.

> push.f {ra, s0-s4, fs0}, -64
addi sp, sp, -64;
sw ra, 56(sp);
sw s0, 48(sp);
sw s1, 40(sp);
sw s2, 32(sp);
sw s3, 24(sp);
sw s4, 16(sp);
fsd fs0, 8(sp)

Yi-Chan

tariqkurd-repo, 2021-01-15:
ok, done!
is it ok now?

yichanfeng, 2021-01-15:
Hi Tariq,

It is OK. 
Many Thanks!

Yi-Chan

closed 2021-01-15 by yichanfeng