riscvarchive/riscv-code-size-reduction #18: RISC-V Table Jump ISA Proposal - C.TBLJAL - initial thoughts/issues.

David-Horner, 2021-01-10:
RISC-V Table Jump ISA Proposal overstates the benefits of C.TBLJAL.

specifically 

> -mrelax
> -mno-relax
> 
>   Take advantage of linker relaxations to reduce the number of instructions required to materialize symbol addresses. The default is to take advantage of linker relaxations.

A) can reduce to C.jal for RV32 (the target for most of our ideas) C.J and C.JAL have an 11 bit immediate and can therefore target a ±2 KiB range. This does not require a CSR setup. it does not require a linker table set up. Thus it is potentially smaller than the proposed mechanism. 
However, if ±2 KiB range is insufficient, the linkage editor can set the c.jal to address a branch to the required routine, either  within a similar table structure, or more significantly to another code segment that also needed to be set up with a larger range, such as:

B) mrelax can change the general form to a single JAL with a ±1 MB range. For one off calls this also uses less space than the  C.TBLJAL as there is a minimum 4 bytes address in the branch table and then the 2 byte C.TBLJAL instruction itself. The benefit point is 3 invocations. In that the standard J formulation can act as a vector for the c.jal mrelax reduction, and can be incorporated after the other active code of the routine for the current routine itself, the point of superior code reduction is unlikely to happen. 

C) even if the target is more than the ±1 MB range, the same option occurs for the generalized forms, use a C.JAL to jump to the jump code appended to this routine, and use that jump code for all other c.jal or JAL that are within range, but not within the original target range.

The point here is that the linker/loader can optimize away the benefits of this specific proposal.

Further, I am about to propose enhancements to -msave-restore that will further negate the perceived  generalized call overhead.

Even if C.TBLJAL were to prove beneficial in some specific configuration, C.TBLJ is not at all required from a code-space reduction benefit as C.TBLJAL can be used instead, with the ra thrown away.. Yes performance may suffer due to return address stack skew, but conventions can be established for the values of the table index in conjuction with the CSR setup to avoid the ra stack push.

 if C.TBLJAL were to prove beneficial in some specific configuration, without significant loss of generality a minimum of  one bit of the C.TBLJAL instruction address can be added to the 7bit literal. This would give a 128 branch address aperture into a 256 branch table. [more bits from the instruction address would give the same size apeture into a much larger branch table]. 
This modification would restore some measure of an attribute of JAL [as described in __Ode to JAL__ pdf attached to email Re: [RISC-V] [tech-code-size] proposal: CALL instruction for LTO, ]
 namely point 4) -  Its target space is not fixed! Rather, depending upon where it is executed, a different range of targets
are available!!

I DO however continue to advocate to for CALLVEC which could implement this table vectoring method, but is inherently more flexible as it relies on the linker/loader to assess the machines specific mechanisms and optimize accordingly.
The C.TBLJAL format could be a compressed form of the proposed CALLVEC that currently maps to SYSTEM opcode space.

tariqkurd-repo, 2021-01-11:
Hi David,

Table jump is broadly similar to one part of Andes Codense, which SEGGER tell me makes a huge difference to code size when they use it in the linker. I agree that the GCC linker can definitely be improved without this, but I'm advised that this technique can take us much further by an expert in RISC-V linker implementation (Paul Curtis). In this respect this is a tried and tested approach which already has production quality software support, although the challenge (with whatever we do) is to get the implementation in the GCC linker.

I'm much more comfortable keeping the J and JAL variants. We *could* say that entries below a certain threshold are J whereas others are JAL, and put the threshold in a CSR so it's not restrictive. We'll save encoding space that way. In fact in that way we can also allow some entries to link to x5, so we could divide into three sections. I'll add some notes into the spec, we should definitely save the encoding space.

David-Horner, 2021-04-16:
re-titled with Pre Sub to be resolved before submission.
Obviously; as no resolution has yet been suggested and the concerns are still outstanding and could easily affect the TBLJMP design.
TBLJMP is very flexible and its application use is greatly isolated from physical implementation details because of the AEE management. 
However, crippling of that flexibility and lack of "emulation alternatives" are not easily corrected after ratification.

closed 2021-07-16 by tariqkurd-repo