riscvarchive/riscv-code-size-reduction #19: If we are still brain storming  .... "Zero page" relocation

David-Horner, 2021-01-10:
Three more ideas (related) that extend the current RVI instructions (under csr control if desired).
These especially benefit embedded systems that have only M and U mode.

The first:
 **"Zero page" relocation:**
   S mode can provide virtualized memory mapping via satp, (address translation and protection).
     This allows the low 2k and high 2k memory address to be mapped (within two 4K pages) anywhere in the virtual address space. 
     As a result register zero addressed load/stores and JALR instructions can directly address "global data and/or routines.

  However, embedded systems (even if they also have an S mode) are unlikely to implement satp.
  Further, high" page" and zero "page" addresses are often reserved  in the implementations address space [for purposes such as initialization code].   

  **Lower overhead head "zero register page" user csrs** could allow enhanced satr functionality [relocation based on function] for these existing x0 addressing instructions. 

I envision two csr with 2k resolution offsets, one for JALR with rs1=x0 and the other for load/store with rs1=x0.

Unenhanced, this would provide a relocated 4k JALR address space for routines, and a relocated 4k "global" data area for load stores.

Details to be worked out are the standard housekeeping: 
- which physical csrs? WARL?
- why user? can S or M mode restrict U mode write? [do we also need to virtualize U mode read for Hyper, etc.]
- assume bits XLEN-1 to 11 define the offset, what other controls go into the low 11 bits? how to encode?
- etc.

tariqkurd-repo, 2021-01-11:
Hi David, at the moment I'm proposing GP/TP relative load/stores with 32-bit encodings so that the data accesses can be shrunk nicely (Anders suggested also using TP) so we can point one to RAM and one to ROM. Clearly only if there's no thread local storage on a core.

It's an interesting proposal to have a GP for JALRs so they can easily reach the code with a short instruction sequence. 

so: JALR x0, offset(x0) wouldn't jump to address "offset" but to JALR_BASE_CSR+offset if JALR_BASE_CSR is implemented.

That should be backwards compatible.

I don't know if 4K is enough to store the commonly accessed functions, and also it may well be too restrictive as in small embedded systems as the functions will probably be split between ROM and RAM so we'd really need two of these to have a big impact, which the table call allows as it's flexible. The linker can move code around in the same section, but not move between sections.

allenjbaum, 2021-01-16:
There may be  a lot of SW side effects if null pointer dereferences are suddenly used for something elsee....

David-Horner, 2021-01-16:
I think this is also a tech list item ...

>
> There may be a lot of SW side effects if null pointer dereferences are 
> suddenly used for something elsee....
yes, there would if it were but it won't.

To avoid all confusion the instruction-apparent zero address (e.g. lw 
rd,(0)0 and sw rs2,0(0)) could be "reserved" from software use.
One data point out of 1024 is a small penalty to avoid the confusion you 
describe. [and 1 in 4096 if #21 were implemented].

However, the instruction does not dynamically translate from pointer values.

Even if a JIT compiler were to use this instruction it would either
1) see the address space as distinct [like x86 IO space], or
2) if it knew the csr mapped address for the "zero page relocation" then 
it would just consider it that part of the full address space.

There is virtually no possibility that null pointers would be mistaken 
for "zero address" formulation.

No more than a lw x1, -53(x2) is a null pointer when x2 = 53.

Null pointer and "zero page" addressing are two entirely different things.

David-Horner, 2021-01-16:
@tariqkurd-repo 
  > To address the concern of a rom and ram memory split, we can add another set of csr registers.

   If the second set is non-zero, then the negative offsets in the rs1=0 instructions (jalr, load/store) use that base address for the "rom" address set. [of course there is no need for the second set to actually be rom, it could be any other mapping that would be useful to the system]  

**a further item from an email:**

>     There was a whole section about having more than a +/- 2K range,  how's that work?
In this proposal there are **two ranges of 4k zero pages, one for code and one for data** with two csr, a csr for each.

**Zero code page:**
a jalr with zero target address register (rs1=0) would branch to the code "page" addressed by the code zero page csr.

That is the basic proposal for code.
However, there are possible enhancements described in #20.

**Zero data page**
load/store with zero rs1 generate an effective address of (ADDRESS-ENCODED-IN-CODE-CSR + INSTRUCTION-OFFSET)

>     Were you shifting the offset? 

Yes. The idea is that, like with c.lwsp and c.swsp, a single positive offset is conceptually much cleaner.

The csr 'page' offset could be offset by +2k and the instructions addresses would thus all look like positive offsets.

However, it may be better to use a pseudo-instruction and make it an abstraction, keeping the [apparent] instruction format the same.

I could go either way. (Not so good for a proposal to include ambiguities).

>   [or] Using

#21 discusses increasing the effective range by using ...

>     the LSBs as upper bits?

allenjbaum, 2021-01-17:
Just to be sure,  do you mean:
   a jalr with zero target address register (rs1=0)
or:
    a jalr with zero target address register (rs1=x0)
?
Having to fetch the register and check that it is zero may affect branch
performance
In any case: this is a perfectly valid instruction now, and this would be
backwards incompatible, except you can make the CSR be zero.

David-Horner, 2021-01-17:
> Just to be sure, do you mean:
> a jalr with zero target address register (rs1=0)
> or:
> a jalr with zero target address register (rs1=x0)
(rs1=x0) , sorry, I was sloppy.
> ?
> Having to fetch the register and check that it is zero may affect branch
> performance
> In any case: this is a perfectly valid instruction now, and this would be
> backwards incompatible, except you can make the CSR be zero.
yes. zero csr for both would be the base RVI situation.

brucehoult, 2021-01-19:
> There is virtually no possibility that null pointers would be mistaken for "zero address" formulation.
> No more than a lw x1, -53(x2) is a null pointer when x2 = 53.

I disagree with that.

Given any reasonable (average size or a bit more) struct S with a member A at offset 0 and M at offset 53, you certainly want p->M to trap as a null pointer dereference if p is 0.

What is more, if the user does q = &p->M (which compiles to an add) then later you want (S*)((char*)M - offsetof(S, M))->A (which hopefully is your -53(x2)) then that also is a null pointer dereference. This looks a bit contrived in C, but it can easily happen in C++.

It's very valuable to have the first 64 or 256 or even 4k bytes in the address space all treated as a null pointer dereference.

I'm assuming here that you meant x2 as an arbitrary non-x0 register, rather than its conventional use as SP in certain ABIs.

However this is completely different than adding new meaning to load/store/jalr when rs1 == x0.

User pointers such as p and q in a high level language can never be allocated to x0.

Aarch64 similarly treats their Zero register (x31) differently depending on what kind of instruction is executing. For most instructions x31 reads as zero and writes are ignored. x31 used as a base register (but not I think as an index register) in a load or store uses the SP. Somewhat weirdly, ADD (immediate) is also an exception, and you can use ADD (immediate) to adjust the stack pointer by referencing x31 as Rd and/or Rn.

brucehoult, 2021-01-20:
Rather than having the base for Zero Page held in yet another CSR, would it not be easier to simply implement x0 as an actual register? Then you can just control muxing it with 0 on read, and enabling write only for some very specific standard instruction.

Using LUI with rd = 0 to set the Zero Page base would be an interesting choice, as we do actually want the value to be constrained to be page-aligned. I imagine most programs would be fine with a compile-time constant for the location of the Zero Page. If there is a use-case for a dynamically calculated value then it's always possible to use runtime code generation of the LUI instruction :-) :-) I doubt it would be changed often enough to be performance-critical.

Otherwise, mv with rd = 0 (aka addi with immed = 0) would work. Unlike the case of Aarch64 x31/SP there is no need to make it specially easy to adjust it by 16 or 32 bytes.

allenjbaum, 2021-01-20:
I think the difference is whether R(base)=0 so the contents of x(rs1)==0,
or base=0 (so, rs1==x0).
I thought he said the latter, so you can distinguish the cases. It's also
much more implementation friendly.

brucehoult, 2021-01-20:
How is that different to what I said?

allenjbaum, 2021-01-20:
Actuall;y -  I misread your statement to be the exact opposite of what you
intended. I need to slow down and read a bit more carefully.

David-Horner, 2021-04-16:
I have re-titled this Post Sub for after the initial submission.
If this gets into the first cut, fine, but it can be done next phase or stand alone.
It does not materially affect any of the other proposals.

closed 2021-07-16 by tariqkurd-repo