riscvarchive/riscv-code-size-reduction #20: Enhanced JALR for embedded, especially when rs1=x0

David-Horner, 2021-01-11:
Enhanced JALR:
Whether in conjunction with "x0 page proposal" issue #19 ,  the JALR rs1=x0  [or possibly more generally for any rs1], bit 20 of the instruction (the lowest bit of the offset) is used to extend the JALR offset:
a) to change the effective granularity from two bytes [an instruction packet] to eight bytes, thus extending by 4 its range and
b) incorporate the 3 bits in instruction format locations 20 to 22 into the resultant target address.

Unlike JALR with rs1<>x0, there is no functional difference in setting the low bit of the offset. Compilers have little incentive to use this bit as a "marker" and neither can hand-coders do fancy tricks based on the low bit in the rs1 register, as x0 low bit is always zero.

Further, as x0 is always zero the additional time to mangle the bits for the new resultant address is not additive to the usual add to register rs1. So, this should not in most designs result in critical path slow down.

Proposed resultant address will be zero extended. when bit 20 is 1.

Details to be worked out are the standard housekeeping:
- how to mangle the bits to achieve the extended address range.
- should the extended range have increasingly larger granularity? 
  - eg. after the 2k at 2 bytes, the next 2k is at 4 bytes, the next 8k at 8 bytes and the remaining 8k at 16 bytes? 
  - [for a total routine space of 22k]
  - or should the smallest new granularity be 8 bytes? 
  - The branch density is not the selling factor, but rather a far call support.
  - any resultant code density is mitigated by the link/loader routine placement.
- should these details also be implementation defined, and effectively be a LTO enhancement?

tariqkurd-repo, 2021-01-11:
very interesting - I agree that it's probably more useful to extend the range - but isn't this is in direct conflict with normal JALR because, or would it only be enabled if the CSR is implemented (https://github.com/riscv/riscv-code-size-reduction/issues/19#issuecomment-758071911)?
I don't think slightly coarser alignment on the target is going to cause a problem.

David-Horner, 2021-04-16:
I have re-titled this Post Sub for after the initial submission.
If this gets into the first cut, fine, but it can be done next phase or stand alone.
It does not materially affect any of the other proposals.

closed 2021-07-16 by tariqkurd-repo