riscvarchive/riscv-code-size-reduction #21: Enhanced lh,lw,ld sh.sw.sd especially when rd1=x0

David-Horner, 2021-01-11:
The third enhancement mentioned in #19, still brain storming ....

Most implementations do not support misaligned load/store.

Especially for load and store when rs1=x0 and "zero page is in effect", as these variants are arguably backward compatible, remove the misalign capability and extend the half, word and double variants to use the low bits to extend the addressing range.
The 4k centred around "zero, wherever that might be" has byte,half,word and double aligned access as usual.

This is critical as it is backward compatible.
 
The next 4k (whether split on either side or not, depending upon mapping of the lower bits) will only contain half, word and double aligned elements.
The next 8k will only contain word and double, and
the last 16K will be load/stores to doubles.

The original 4k addressability has expanded to 16K for RV32 integers and single float , and to 32K for RV64 integers and double float.

Once the toolchain has accommodated this extension (and also true for the #20 enhancement) for the rs1=x0 case, expanding to rs1<>x0 is a minor step. 

Note, systems that handle misaligned elements through emulation can as easily switch over to emulating this expanded addressability.

Looking forward, Tariq, to hear your response.

tariqkurd-repo, 2021-01-15:
Very interesting (sorry for the delayed reply). Are you sure it's backwards compatible? It would mean not taking the misaligned fault and doing useful work instead, which sounds fine to be but only if we cna get away with it....

tovine, 2021-01-15:
Could this not be defined in a new extension? And if so that extension would have the freedom to define this behavior, just like we take freedoms to undefine the floating point load/store ops in Zfinx

David-Horner, 2021-04-16:
I have re-titled this Post Sub for after the initial submission.
If this gets into the first cut, fine, but it can be done next phase or stand alone as per @tovine 
It does not materially affect any of the other proposals.

closed 2021-07-16 by tariqkurd-repo