riscvarchive/riscv-code-size-reduction #22: Push pseudo code sp adjustment location

chuanhua, 2021-01-15:
In 4.1, pseudo code of push instruction, the sp modification statement should be put at the end after all of the store instructions. So even if an interrupt/exception happens, the sp will not be modified.

tariqkurd-repo, 2021-01-15:
Hi @chuanhua, the problem here is whether the interrupt handler allocates memory on the stack using SP.  If it does then PUSH should increment first and POP should decrement last so that the stack doesn't get corrupted. This, of course, make the interrupt/restart sequence problematic because (as you say) we can only increment the stack pointer once. We can issue the load/stores more than once but not the SP increment, or the RET for POPRET.
We can't claim that the sequences are uninterruptable, because there could be a bus error during them.

The idea was to show the overall behaviour with the pseudo-code and I'll add a section on implementation.

tariqkurd-repo, 2021-01-15:
I've updated the doc using a recommendation from Andrew Waterman about the software view, from an email discussion last year. Let me know what you think.

chuanhua, 2021-01-18:
I have reviewed "section 9. software view of implementation". The description looks fine to me. However, I still think that it will be clearer to put the SP modification statement at the end of the PUSH pseudo code.

I do not understand the stack corruption problem if an interrupt handler wants to use the stack in your reply. Even if an interrupt handler has used the stack to allocate memory, when it returns, the SP will get its original value before the interrupt and the PUSH instruction will be executed again to store the specified registers. Could you elaborate more?

tariqkurd-repo, 2021-01-18:
You're right - I've moved it to the end. It makes more sense as it matches the software view.
Please check it.

yichanfeng, 2021-01-19:
Hi Tariq,

For the latest change of stack adjustment, it seems that the first store address (variable **"addr"**) is wrong.
-> expected: addr = sp (previous pseudo code is correct)
-> got:          addr = sp+stack_adjustment (latest pseudo code)

> ///////before
addr=sp;
sp+=stack_adjustment; //decrement
//////after
addr=sp+stack_adjustment;

Yi-Chan

tariqkurd-repo, 2021-01-19:
You're right Yi-chan, it was a mistake. I've fixed it now. Please check it, and thanks for your help.

yichanfeng, 2021-01-19:
Hi Tariq,

The address calculation is correct in the latest PUSH pseudo code. 

Many thanks!

Yi-Chan

yichanfeng, 2021-01-19:
Hi Tariq,

Sorry, I have one more question:  
In the following sentence, is the description about "the reserved encoding for spimm" should be changed from "spimm[1]=0" to "spimm[2:1]=0" since the encoding table (Table 2. proposed 16-bit encodings) shows that **"C.POP[4:3] = 2'b00"** ?

> For C.POP, spimm[1]=0, and rcount[2]=0 are reserved, as these encodings give minimal benefit

Yi-Chan

tariqkurd-repo, 2021-01-19:
ok fixed.

tariqkurd-repo, 2021-01-20:
Hi @chuanhua can I close this issue now?

Tariq

closed 2021-01-21 by tariqkurd-repo