riscvarchive/riscv-code-size-reduction #24: Towards quantifying Optimization: guidelines and principles.  - None without disruption

David-Horner, 2021-02-03:
##  **Optimizations by definition disrupt the norm.**

 Whether it be convention, or construction, there is modification from the current standard with side-effects related to each optimization.

 ### Almost always the optimization entails some up front enabling overhead.

 Thereafter, the primary pain is its non-obvious nature, until such becomes the norm, and anticipated.

 ### The best optimizations are those that generalize a specific class and have multiple means to implement to tailor to the specific situation.

 We are all familiar with this in the ISA field:

 The same ISA can [and RISCV is especially able to] be implemented with many physical designs.

- FPGAs, handwired ASIC, High Level Chip Generators

-  single to 12+ stage designs.

-  Re-named or “fixed” registers.
 
-  Emulated or hardware implemented instructions or allowed alignment or csrs.
 
-  Single issue, opcode fusion, multiple issue, OoO.
 

 ### These were at a cost.

- “Wasted” X0 encoding.
--        to make decode simpler
-  Coder unfriendly immediate mangling.
--         to reduce chip trace length and decode simpler
-  Jalr Low Order Bit “say-what?” Complexity wart.
--        “If and only if the immediate low order bit is 1 AND the register also has the low bit set will the low order bit contribute to the target address generation, otherwise both bits are ignored and effectively assumed to be zero.”

### The “C” Compressed Code optimization generalized many ideas ; some disrupted the orthogonal RVI design.

- 1 to 1 map of Compressed instruction to underlying instructions.
 
-  Exclude X0 variants and other tool-chain friendly consistency.
 
-  Optimized sub-set of registers.
 

- -  Special purpose specific registers 
- -  --       return address [ra, x1] and stack pointer [sp, x2]
- -  --        Only x8 through x15 encoded in many instructions. 

- RV32C only and RV64C only instructions 

### Some of these decisions/generalizations will need to be abandoned to further optimize code.

#### Some mistakes in hindsight would have been avoided:

 - RV32 and RV64 common encoding for W [word] operations.

- JALR should have always cleared the low bit of rs1 and the immediate low bit 1 encoding been reserved 

### Premature optimizations - 
-   Although the C instruction set was based on empirical data, that data was the result of under optimized code.
- --   ADDIW can be optimized away in most index related code, its then highest frequency use.
- --  However, C.ADDIW displaced C.JAL because link/loader optimized placement and jump-chaining had not yet been implemented.   
- C float load/store variants did not anticipate Zfinx and thus diminishing returns for RV32 code compression. 
-  -- rather C.FL/SD* variants use could supplant single in the future. 
- -- Having the RV32 C.FL/SS* encoding reserved would have allowed future flexibility for the scarce 16bit encoding space.

### Similarly, we should look at proposed ISA "optimization" instructions with an eye to future software obsoleting and with mechanisms to recover their encoding if the future reveals we made an error in allocation.

## Software also needs to look beyond what we currently do in tool-chain 
  We must envision means to leverage RISCV's strengths.
  A new mechanism and enhanced infrastructure may be needed to make significant advances in code size reduction AND code speed up gains.
  RISCV is uniquely suited to this purpose with its elegant low cost design. 
     E,g. The vision that RVI can replace microcode has not yet been realized. 
  Providing new mechanisms in the ISA that the tool-chain can cooperatively optimize, rather than niche, and especially macro, instructions, will advance our cause best.

tariqkurd-repo, 2021-02-05:
>E,g. The vision that RVI can replace microcode has not yet been realized.
>Providing new mechanisms in the ISA that the tool-chain can cooperatively optimize, rather than niche, and especially macro, >instructions, will advance our cause best.

Why doesn't table jump achieve this? It's calling a routine which can be considered micro-code. Maybe I haven't figured out what you mean by micro-code.

David-Horner, 2021-02-05:
The vision that RVI base instructions can replace microcode.

This does NOT mean micro-code/micro-sequencer implementations of complex instructions.

But rather, just as memory misalignment support can be emulated, complex instructions have efficient, very low overhead, mechanisms to support "emulation".

Note: the microcode is not restricted to RVI instructions, but the base instructions are all that are required to use the mechanism.

       Specifically, it does

1) not rely on special instructions within the micro-code segments,

2) nor rely on special locations nor special architectural states

             [beyond the standard mechanisms such as csrs, including informationreported in xstats].

As a result, the special jump instruction proposed for the xtvec branch table would not qualify.

   [This was proposed early on in fast-interrupt TG by Krste, and I supported it. I might still support such a mechanism but it is not "ideal"].

The key phrase is rely on. Facilities outside of the micro-code itself that accelerate the micro-code are definitely part of what I envision and am proposing.

For example,

a) special caching of micro-code,

b) pre-decoding of the micro-code by either hardware alone or in conjunction with software in the Application Execution Environment layer.
>
> It's calling a routine which can be considered micro-code.

I would not consider the original TBLJMP "call" mechanism micro-coded, nor even micro-code enhancing.

It targets the standard ABI call to standard subroutines.

However, the "emulation mode" of TBLJMP is the kind of RVI instruction micro-code enhancement that I "envision" and why  I proposed it.

[Also, why I proposed the very similar CALLVEC].

> Maybe I haven't figured out what you mean by micro-code

I'm sorry I left so much to be figured out.

Hopeful I have explained enough now that the ideas is clear.
>

David-Horner, 2021-04-16:
re-titled with Pre Sub to be resolved before submission.
Obviously; as no resolution has yet been suggested and the concerns are still outstanding and could easily affect any of the proposed features.

closed 2021-07-16 by tariqkurd-repo