riscvarchive/riscv-code-size-reduction #26: Reason for reserving rcount[2]=1 && spimm[2]=0 for C.PUSH/C.POPRET

chuanhua, 2021-02-24:
What is the reason for reserving rcount[2]=1 && spimm[2]=0 for C.PUSH/C.POPRET?
It seems to say that the more the registers are saved/restored, the more the extra stack adjustment (>3).

tariqkurd-repo, 2021-02-24:
Hi @chuanhua this was an attempt to save encoding space, but it does seem a bit awkward. I've improved the wording, its:

rcount[2]=1 && spimm[2]=1 that's reserved

Most of the uses of C.PUSH/C.POPRET have rcount < 4 (i.e. rcount[2]=0), and thd cases where rcount[2]=1 typically have spimm[2]=0, and so restricting the range of spimm for rcount[2]=1 doesn't make a huge different to the code-size.

Does you think this kind of rule is sensible to reduce the encoding space?

tariqkurd-repo, 2021-02-26:
I've clarified the specso I'll close this

closed 2021-02-26 by tariqkurd-repo