riscvarchive/riscv-code-size-reduction #28: How to move along code reduction related ideas

David-Horner, 2021-03-01:
I have made many proposals for improving support for reducing code size.

However, I have little guidance on how to move these along.

I outlined a plan to revise the tool-chain to
a) improve msave-restore 
b) for RVI reduce call size by "jump-chaining".

I proposed augmenting definition of macro-instructions [such as TBLJMP and PUSH/POP] to allow efficient simulation of these instructions, specifically to ensure that the emulation does not need to save/restore registers over what the compiler will reserve for the macro-instruction. This is in keeping with the RVI philosophy of using M-mode as "micro-code", by extending gthe concept to AEE.

I proposed encoding new-standard/custom extensions as 64bit [and by extension to 80bit] instructions that map to routines to emulate, or to the standard or vendor specific encoding.
Related is the proposal today to specifically define [initially primarily] fusible instruction sequences that can be replaced by instructions that perform the sequence as a custom instruction in the reserved custom instruction space. 

There are various other suggestions [I and others made] that now appear to fit into the category of "Not dealt with at the TG level".

If not raised initially here, and then escalated to where they "really" belong, how are these ideas to get attention?
What is the mechanism to present "higher order" ideas?
How does someone like myself get guidance on how to present such ideas and move them forward?

tariqkurd-repo, 2021-03-02:
these are really good questions - I think emailing them to toolchains&runtimes is probably right.

tech-toolchain-runtime@lists.riscv.org

it's up to them to improve the toolchain, and create TGs to do so if necessary
these are all very valuable pieces of work, we need to make sure they actually go somewhere.

David-Horner, 2021-04-16:
re-titled with Pre Sub to be resolved before submission.
Obviously; as no resolution has yet been suggested and the concerns are still outstanding and could easily affect the TBLJMP design.
We are not developing enhancements in a vacuum ....

closed 2021-07-16 by tariqkurd-repo