riscvarchive/riscv-code-size-reduction #32: No Need for sext.w

RogierBrussee, 2021-04-03:
Sign extension from 32 bit is supposed to be ADDI rd rs1 0 which for rd == rs1 can already be done with C.ADDI rd  0 (as Is, if IIRC, even mentioned in the C spec).  In any case there is no need for a separate  sext.w instruction.

On a related note, zext.w requires Zba not just Zbb (https://github.com/riscv/riscv-bitmanip/issues/120).

David-Horner, 2021-04-03:
>
> Sign extension from 32 bit is supposed to be ADDI rd rs1 0 which for 
> rd == rs1 can already be done with C.ADDI rd 0
I believe it is ADDW rd rs1 r0 that will do.
>
> (as Is, if IIRC, even mentioned in the C spec).
[The C spec would need instructions in the ISA to effect the behaviour]
>
> In any case there is no need for a separate sext.w instruction.
I agree 100%
>
> On a related note, zext.w requires Zba not just Zbb 
> (riscv/riscv-bitmanip#120 
> <https://github.com/riscv/riscv-bitmanip/issues/120>).
But this is where the problem is.

Ken has announced 
https://github.com/riscv/riscv-bitmanip/issues/121#issuecomment-812695424

that The Zb[abcs] instructions, encodings, and mnemonics have already 
been approved.

So, the argument is that this change request is too late.

It destroys the symmetry with the other sext instructions sext.b and 
sext.h.

But more to the point, for symmetry, why is there no addb rd rs1,rs2 and 
addh rd rs1,rs2?

When these are added the sect.b and sext.h become redundant.

Unless of course the r0 variants are reserved, but that goes in the face 
of the current "common sense" symmetry redundancy.

Reserving redundant formats, presumably, "would also make the code much 
harder to read, write, and debug "

Apparently we cannot have that, but rather consume opcodes now :

> providing new standard instructions that reduce the code footprint 
> while increasing performance
For "standard" read 32bit ops from the "reserved" op code space.

Ken does not mean 48bit or 64bit instructions  even if they would meet 
the footprint and performance objectives [as many bit-op would].

I suspect the reason 48bit and 64bit have been excluded is the same for 
every group.

They want first pick on the choice op-codes. Who wouldn't?

The question still unanswered, and perhaps not forcefully enough asked 
is " Why should bit-op get them?".

Ken weighed in on this at 
https://github.com/riscv/riscv-bitmanip/issues/121#issuecomment-812695424

I ask the further question

"Why squander *reserved* op-codes? Why use them for "optimizations" of 
operations that are definable as a sequence of 2,3 or 4 instructions?

"The vendors custom space can be readily used for that.

"Why not exclusively use reserved opcodes for operations that are [in 
their nature] conditional, iterative or in excess of 4 "standard" 
instructions?"

We can always raise issues during the Public Review phase.

Best not to rise them here, because during public review people may ask,

      "if you knew this was an open question, why are you putting it 
forward as a pending ratification proposal?

       Does RISCV.org not seek consensus?"

Issues/questions raised during Public Review,  except that kind, can be 
addressed by a pseudo-technical explanations that are not fully reasoned.

But the policy questioning require a logical, reasonable, common-sense 
and non-technical response that actually addresses the concern.

Then again , like the already approved _Zb[abcs] instructions, 
encodings, and mnemonics_ — it may now be too late to keep this genie in 
the bottle.

RogierBrussee, 2021-04-03:
@David Horner I actually meant that there is no need for C.SEXT.W in this proposal (and you seem to agree).  

I argued long ago that support for shorts and bytes was necessary and that add.b, add.bu, add.h, add.hu were preferable to sext.b, zext.b, sext.h, sext.hu but they do take a bit more opcode space  and C   suggests that all arithmetic can be done as 32bit ints. So Claire preferred just sign and zero extension instructions <shrug>. 

I do think at least the Zbb, Zba, Zbs, the pack instructions and likely Zbc instruction are perfectly reasonable as reserved 32 bit instructions: IMHO they are a bit misnamed as _bitmanip_ instructions and should really be thought of as (rather useful) completions of the basic general purpose _integer_ instructions that will routinely used by the compiler in a few years and change RV from a spartan to a lean instruction set, and I believe IMZbbZbaZbcC (+ Zce?) will be the de facto minimum configuration for a CPU.  The fact that some of the instructions show up here in code compression goes to show that.

tariqkurd-repo, 2021-04-06:
Very interesting - you're right SEXT.W is listed in the umprivileged spec as as pseudo-instruction for addiw rd, rs, 0. thanks for bringing this to my attention.

It is *extremely* annoying to have to include Zbp to get pack for ZEXT.W, whereas everything else Zce needs is covered by Zbb.
Seeing as SEXT.W is being removed, I'll also drop ZEXT.W. I don't have any benchmarking results for it yet anyway, and if someone wants it then they are welcome to also include Zba. I'll put a note in the spec accordingly.

closed 2021-04-06 by tariqkurd-repo

RogierBrussee, 2021-04-06:
As I wrote, for ZEXT.W. there is no  need for  Zbp (the fancy bit permutation extension), just for Zba  (the address computation extension which you were already contemplating).  While the point of Zba could be considered code compressing, and a good fit wit Zce (including for RV32), I agree that they should have just included SLLI.WU or at least ZEXT.W in Zbb. Your complaint probably carries more weight than mine.

tariqkurd-repo, 2021-04-06:
it's worth a try, it might be too late to change the subsets in Bitmanip but you never know