riscvarchive/riscv-code-size-reduction #33: Rationale for CSR-based jump tables

irdc, 2021-04-14:
Looking through the discussions of the code size reduction group, I’m having trouble finding the rationale for CSR-based jump tables (TBLJAL). I’m curious as to why GPR-relative jump tables were rejected, as something like:

    ldjalr x1, 0x1234(x2) # reg[1] := pc +1; pc := mem[reg[2] + 0x1234]

looks so much more convenient to me:
* more general
* no CSR to save/restore upon task switch, so no changes to (RT)OS needed
* can use multiple jump tables

Of course you’d be sacrificing a GPR; though gp (x3) looks like it might be a convenient candidate that wouldn’t have this problem.

abukharmeh, 2021-04-14:
As far as I know, Its all about the encoding space used to encode such instruction, and any additional instruction you need to execute that instruction.  Each register specifier is 5 bit for the whole range, or 3 bits for the compressed range. 

I am not sure how your propossed instruction would have any code size savings over  normal JALR ?

irdc, 2021-04-15:
That depends on the exact operation being encoded by the compressed instruction. Assuming rd=ra and rs1=gp, the following instructions would be equivalent:

    c.ljal 0x12
    ljal ra, 0x12(gp)

Which would leave the same 8 bits for the displacement. The 32-bit encoding of a hypothetical LJAL could have the same format as an ordinary load instruction; that is, with a 12-bit displacement.

tariqkurd-repo, 2021-04-15:
Hi @irdc we can't really allocate a standard register to this, without creating a new one for this purpose. As @abukharmeh pointed out we can't encode one in the 16-bit encoding and forcing the use of a fixed register is generally a bad idea - if we use GP then we lose the ability to load/store via GP. TP is a better option but we already have a plan for that to use it as a second TP where there is no thread local storage (software change only) and I have heard other plans for TP in the past too, we can't allocate it to us for this purpose. The current TBLJAL spec makes no assumptions about the use of registers, and it also avoids the need to save and restore the base of the table when changing privilege levels as there is a CSR per level, so when jumping to the interrupt handler and back to the same task then no save/restore is required. If we used a GPR then we would have to save/restore it each time assuming the interrupt handler has a different table setup.

Reusing a GPR really means taking away with one hand which you give with the other, we lose the ability to use that GPR and so code size will inevitably increase.

The main use for TBLJAL is for small embedded platforms where there is only one table set up for the whole application and no dynamic loading, so I don't expect the CSR to need to be saved/restored.

I don't see why using a CSR limits the ability of the instruction, because if you can update the table base in a GPR then you can update it in a CSR in a very similar way but without the penalty of losing a GPR for this purpose.

tariqkurd-repo, 2021-04-15:
More generally you're asking for a new ABI for TBLJAL which is required to allocate a dedicated GPR for this purpose, so it would be incompatible with the existing raitified UABI and proposed EABI - we'd need an additional version of each. Using a CSR avoids this. I hope that clarifies the situation.

closed 2021-04-15 by tariqkurd-repo

David-Horner, 2021-04-15:
TBLJMP was discussed extensively on the list. [My next post will have a link to various threads]

To address your context switch concern:
As @tariqkurd-repo points out CSRs for each user level are defined.
But further to that, the address for the U-mode [and S-mode if it exists] csr pointer is virtual.
That means applications running in U with satp enabled will use virtual address for the jump table that can be the same for all applications, but point to different physical addresses and hence different branch vectors, all under the control of the AEE - Application Execution Environment - that is the OS when Linux is running, and usually S-mode , however satp can be active even if all management is via m-mode.

The most significant aspect of this feature is that the application can use it without any awareness of the location on the table or its maintenance.
This level of abstraction is very valuable and consistent with the philosophy behind virtualisation which RISCV has designed in from the beginning, especially the concept of the AEE and other Execution Environments.

As I mentioned above, I am compiling a list of entry points to the threads discussing TBLJMP on the list and hope to post it soon. [but I am rushing to eat here as sunrise is almost here in Toronto]..
 u