riscvarchive/riscv-code-size-reduction #34: Anything wrong with RVM22? Is there 22 bit for RISC-V?

Lucas-Wye, 2021-04-16:
https://github.com/riscv/riscv-code-size-reduction/blob/master/ISA%20proposals/Huawei/Zce_spec.adoc#zce-configurations#64

```
* `Zcea` is required in RVM22, is compatible with all existing standard extensions
** `Zcee` is a subset of `Zcea` which required in RVA22
** `Zcea/Zcee` imply `Zba/Zba` to get the 32-bit versions of the sign/zero-extension encodings
* `Zceb` is optional for in RVM22 _because it conflicts with the D-extension_
** `Zceb` is fully compatible with `ZDinx`

```

tariqkurd-repo, 2021-04-16:
Hi Lucas - RVM22 is the 2022 embedded platform they will include Zce. There's also RVA22 for the linux platform.
It's nothing to do with 22bit.

Tariq

closed 2021-04-16 by tariqkurd-repo

David-Horner, 2021-04-16:
That is M for mobile and A for application, correct?

Is there another category?
Where is the link to the specification and definitions?

tariqkurd-repo, 2021-04-16:
I think it's A for application, there's no current official spec as far as I know, and just those two categories. RVM22, RVA22.

jim-wilson, 2021-04-16:
The 22 is the year, i.e. 2022, which means the spec can only contain extensions ratified before that year (or maybe that year or before).  There are RVM19, RVA19, RVM22, and RVA22 architecture profile proposals.  The first two won't include any of the extensions raitified last year or this year, the last two will.  I haven't seen any formal definition either, but they were mentioned in a talk that Krste gave.