riscvarchive/riscv-code-size-reduction #36: maybe use BRANCH major op, for branch instructions

jnk0le, 2021-04-18:
It would simplify decoding and branch prediction. (maybe i'm wrong)

There are 2 empty `funct3` slots in BRANCH major op and current encoding of DECBGEZ,BEQI,BNEI allows to put all of them in only one slot.

They will also move to 'D' independent zce*** subset.

tovine, 2021-04-19:
In a way that makes sense, although the instruction encoding format is quite different so that might cause more complexity in the decoder if one opcode has two very different ways of encoding parameters?
Also, I believe the encoding was chosen because the compressed version conflicts with the C.FLD anyway, but if you have any compelling use cases that need this instruction and double precision floating point you're welcome to state their case and we can look at it in the next meeting (which is tomorrow)

tariqkurd-repo, 2021-04-19:
Hi - two points here:

1. As @tovine said, the DECBGEZ encoding is chosen so that it uses a D-extension encoding, as the 16-bit form requires a D-extension encoding and it's much less confusing if both are in the same subset.
2. I agree that it would probably make sense to group BNEI/BEQI with other branches, unless some other extension has allocated those opcodes. Therefore at the moment the BNEI/BEQI encodings are in the CUSTOM-0 space. However, it does seem that they would fit in that encoding group so I'll put a note in the spec suggesting that they go there. It's not very clear how to allocate opcodes to new extensions, so I think that this is the best approach for now. The spec will need to go through an "opcode and consistency" review before public review, and at that stage will will agree final encodings.

is that ok? Can you close the issue?

closed 2021-04-19 by jnk0le