riscvarchive/riscv-code-size-reduction #37: why former bitmanip compressed instructions are no longer available for RVA

jnk0le, 2021-04-18:
sign/zero extension as well as `c.not` and `c.neg` were removed from B ext because because it will be handled by code size group, so we can have greater fragmentation.

Now I see that `c.not` and `c.neg` are grouped together with things like `MULIADD`, `TBLAJL` and `PUSH/POP/POPRET` even though there is Zcee with sign/zero extension instructions taht came together with `c.not` and `c.neg`.

What is the reason to not add `c.not` and `c.neg` to Zcee? Especially that those instructions were considered a close to no-cost on big cores.

tariqkurd-repo, 2021-04-19:
That was a decision by Greg Favor and Andrew Waterman, I actually agree with you but I'm not responsible for defining RVA22. I think C.MUL is in the same category, very low cost to implement and has benefit.

closed 2021-04-19 by tariqkurd-repo

David-Horner, 2021-04-19:
Please provide the evidence of Greg and Andrew's decision and deliberations.
Thank you.

tariqkurd-repo, 2021-04-19:
https://lists.riscv.org/g/tech-bitmanip/topic/zext_w_not_in_zbb/81893928?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,81893928

jnk0le, 2021-04-19:
They don't look to be a diehard against `c.not` and `c.neg` inclusion in RVA. `c.mul` will be included in Zcea so the 16bit encoding space will be "wasted" anyway.
Maybe @aswaterman can say more about that decision.

tariqkurd-repo, 2021-04-19:
if @aswaterman wants to include more Zcea instructions then it's fine by me

aswaterman, 2021-04-19:
`c.neg` is an odd one because its expansion doesn't match the pattern of the other instructions that have an rd'/rs1' argument (the RVI negation idiom needs the source argument in rs2, not rs1).

tariqkurd-repo, 2021-04-20:
Yes - I'd noticed that and I've also put a note in the spec saying that it costs additional muxing during the expansion.

jnk0le, 2021-04-21:
`c.neg` could split rs2' from rd' into bits [4:2] where rs2' normally exists. There is some reserved space like of this type but that might be quite wastefull wrt. extra size saving.

Alternatively put rd'/rs2' int bits [4:2]. rd' is already muxed from there in some instructions. But there arises an issue with placing anything else around occupied encoding.

tariqkurd-repo, 2021-04-22:
true, that's possible but then it takes up 8x as much encoding space for minimal extra benefit.

DSHorner, 2021-04-22:
instead the u-impl can directly trigger the ALU to negate, or select the 
alternate path from the register file rather than build the instruction 
per se.

jnk0le, 2021-04-23:
Excluding `c.neg` there is `c.not` and `c.mul` that can be easily put in Zcee.

RogierBrussee, 2021-04-26:
The 7 code points of C.ADDI4SPN with nzimm == 0 except for C."Illegal instruction"   (ie. code points  0x1..0x7) are reserved and have an rd'  at the position of rs2'. Thus, if one uses those codepoints for C.NEG, one can use the existing mux with rs2'  at the cost of only being able to use rd' = x9...x15 (i.e. not x8 = s0) with C.NEG.