riscvarchive/riscv-code-size-reduction #38: move muliadd/adiadd like instructions out of Zcea

jnk0le, 2021-04-18:
Regardless those instructions end up as Claire's c3 format, 48 bit or something more crazy.
I propose to split those instructions out of Zcea int something like Zcec etc. 

In this way the Zcea can be frozen and ratified without big obstacles.
Also not everyone might want to implement Claire's c3 or 48bit formats, but are happy with the rest of Zcea.

tariqkurd-repo, 2021-04-19:
Do you have specific examples of "not everyone" or is this theoretical? 

We can't offer a full range of options to people and a standard toolchain and set of libraries which supports them all.

David-Horner, 2021-04-19:
> We can't offer a full range of options to people and a standard toolchain and set of libraries which support them all.

When can we not? Now? In a year? In two years? 
How long lived do we expect these subsettings and groupings to exist?
If they are just tactical designations that we expect to replace in a year or two, then I agree.
However, if these are fundamental constructs that we expect to last for a generation or generations, then we can indeed expect to have be able to "Please all the people all the time" and support any and all combinations..

tariqkurd-repo, 2021-04-19:
The standard approach here is to allow each instruction to individually exist in the config string. To support such a configuration a custom toolchain would be required, and if a company takes the time to do that they may as well add custom switches to the compiler to enable an arbitrary subset so in fact the config string approach probably isn't required in practice.

More subsets can be added in future, once ratified, but the instruction definitions can't so at this point it's much more important to get the definitions correct.

>We can't offer a full range of options to people and a standard toolchain and set of libraries which support them all.

This statement is the reality we live in.

closed 2021-04-19 by tariqkurd-repo

jnk0le, 2021-04-19:
Now I see that muli is also c2, so there wouldn't be much left in zcea. #39 and #37 (with c.mul) seems to be better than this.