riscvarchive/riscv-code-size-reduction #39: make tbljal and pushpop as an another sub-sub extension

jnk0le, 2021-04-18:
(Talking about official 0.29 draft as the huawei proposals had something like that iirc.)

I propose something like that:
[image]

those two sets of instructions alone contribute the great part of measured codesize saving each, as well as the pushpop provides a lot of performance/power-efficiency gain on single bus cores (like cortex-m0)

It is imaginable that some implementors might want to implement Zcpushpop and maybe Zctbljal (that adds extra CSR state) without the rest of Zcea.

David-Horner, 2021-04-18:
I support this approach.
However, given that we have no guidance on how to sub divide or sub-sub divide or especially, as in this case, cross manage extensions; I expect that whatever arbitrary decision we come up with will be sub-optimal and restrictive for at least one of the intended purposes/expectations [which have only inferred and subjective] of this exercise.

tariqkurd-repo, 2021-04-19:
We spent almost the full hour of the last meeting discussing subsets. There are many many ways to categorise the instructions.

Your suggestions are very valid, and I thank you for them.

From my point of view: small embedded cores need all of Zcea and Zceb. Zceb only exists in case someone wants D and a FP separate register file (which I think is unlikely for an embedded core - I think ZDinx is more likely and to keep Zceb).

TBLJAL is very similar to CLIC *(xTVT). PUSH/POP are very simple on a small core. MULI/MULIADD/ADDIADD all have low impact on area and timing. All will save more area in on-chip memory than they cost in gates for the core.

So my strong recommendation is to implement all of Zcea and Zceb. I'm planning to request that Zcea required for RVM22. Zceb will only be optional for RVM22 in case someone wants D and not ZDinx, however the actual decision is not down to me.