riscvarchive/riscv-code-size-reduction #40: reuse 2*XLEN load stores as load/store pair

jnk0le, 2021-04-19:
(not sure if this should go here or somewhere else)

Proposal is to use 2*XLEN sized load/store instructions (ld/sd on RV32, lq/sq on RV64, RV128 excluded) for loading/storing pair of registers. Currently those encodings are reserved for XLEN matching use.

Most obvious purpose would be the Zdinx and P extension (Zpsfoperand) enabled RV32 cores, So the compilers can "natively" use 2*XLEN sized types. 

Those instructions definition will have to handle the little/big endianess problems (regpair handling vs endianess is liquid at the moment)

LDGP, SDGP, LDTP, SDTP could be also made available on rv32 in the same manner.

Of course this extension will compete with general purpose load/store multiple (as is the case on arm32 already), if that is also planned. 

About potential implementation issues (mainly performance), here is my experience with arm:

##### cortex m3,4
>LDRD and STRD cannot be pipelined with preceding or following instructions. However, the two words are pipelined together. So, this operation requires three cycles when not stalled.

https://developer.arm.com/documentation/ddi0337/h/programmers-model/instruction-set-summary/load-store-timings
LDRD/STRD perform similarly to 2 element LDM/STM.

##### cortex m7
LDRD, STRD can't be dual issued with preceeding or following instructions (even partially, half cycle slidestall turns into 1-2 cycle loss). The core however can dual issue two normal loads if there is no (even/odd) bank conflict. So the two consecutive compressed loads are always better than LDRD. This is not the case for stores probably because it was not considered important in presence of STRD and STM.
LDRD/STRD perform similarly to 2 element LDM/STM.

tariqkurd-repo, 2021-04-19:
I'd view this as more of a performance improvement than a code-size reduction.
It certainly is missing in the architecture, and it should be considered. It's possible in the implementation to detect 2 consecutive load/store words with 16-bit encodings and issue them as a single double word, but there's nothing in the ISA which would make the decode simpler and also change the semnatics by having a single double width load or store.

I think this should be taken on by the code-optimisation SIG, which we haven't managed to get going yet, so I'm not sure where this should be sent. I suggest emailing it to tech-unprivileged@lists.riscv.org

tovine, 2021-04-19:
True, it could be a valid code size improvement if C.LQ had been available for use in RV64, and C.LD in RV32. But those encodings have already been reused by C.FLD and C.FLW respectively.

jnk0le, 2021-04-20:
> True, it could be a valid code size improvement if C.LQ had been available for use in RV64, and C.LD in RV32. But those encodings have already been reused by C.FLD and C.FLW respectively.

c.fld/c.lq and c.fsd/c.fsq are already taken by Zceb.
Zfinx can allow us to similarly repurpose c.flw/c.ld and c.fsw/c.sd for compressed load pair on RV32 only.

tariqkurd-repo, 2021-04-20:
When Krste agreed to reusing the c.fld etc. encodings he was clear that he didn't want the single precision versions reused. 
@jnk0le are you ok to email tech-unprivileged@lists.riscv.org and close this one?

jnk0le, 2021-04-21:
I'm not a member, so can't post to the list. But i'm ok with closing this here.

closed 2021-04-21 by jnk0le