riscvarchive/riscv-code-size-reduction #44: Proposal: compressed Load Program Counter instruction

RogierBrussee, 2021-04-27:
One can use 

AUIPC rd imm_high
LXX rd  imm_low(rd)

for PC relative addressing e.g. for constants of type XX (= BU, B, HU, H, ..). The combination can even be fused. If that constant is stored near (±2k) to the site of the instructions, that AUIPC will have imm_low = 0, however.  To make PC relative addressing cheaper in terms of code size introduce one of the following instructions

#Load Program Counter
C.LPC rd -->  AUIPC rd 0

#Load Program Counter aligned to the previous multiple of 16.             
C.LPCALGN16 rd --> AUIPC rd 0; ANDI rd rd (~15)  

The load program counter instruction C.LPC is, simplest and the alignment can be done with the 32 bit load  instructions (at least, if one does not use the last "unaligned bits" to extend the range).

The  aligned version  C.LPCALGN16 has the advantage that it can be used with the compressed load and store instructions (or the extended range for unaligned bits trade). 

There is room for this instruction with the reserved C.LUI rd nzimm == 0 (i.e. the codepoints with bit01 = 0b01,bit2--6,12 = 0b0 ,bit13..15 =0b011)  with a full 5 bit rd  (excluding rd = zero, and arguably sp), or with the new compact unary instructions with a 3 bit rd and a slight oddness because it does not use rs1. 

Rogier

tariqkurd-repo, 2021-04-27:
Hi @RogierBrussee - I'm not aware that AUIPC ; LH etc. is a big problem for code size.

Most load/stores are word size (or word/double for RV64) which are covered by LWGP/SWGP and the TP versions and the RV64 versions LDGP etc.

Only a very small proportion of load/stores are halves/bytes and I save code size for most of those with the 16-bit proposed encodings C.LBU/C.SB etc.

Do you have a specific application where C.LPC would help a lot (i.e. by at least 0.5%)?

Tariq

RogierBrussee, 2021-04-28:
That you have introduced LWGP/SWGP and friends is a very good point and the compact versions are impossible to beat in terms of code size.  

However, I have repeatedly read that rather than providing instructions for long constants one should just use PC relative addressing and the normal load instructions including the floating point ones for floating point constants.  I understand ARM supports PC relative addressing "natively" and uses it extensively, suggesting that this can be a useful technique. A C.LPC (and even more so, its aligned version) is a way of making PC relative addressing cheap(er) on RV at near zero cost in terms of codepoints and additional logic.  In particular it hopefully reduces the pressure on what can be made easily available through C.LWGP and friends.

tariqkurd-repo, 2021-04-28:
Hi @RogierBrussee,

PC relative loads have come up before, and we're not keen on them for a few reasons because they put constant data in the text section:
1. they cause cache fragmentation (loading instructions into the D-cache)
2. they cause a security risk - it means the text section needs read access so it can be scanned for ROP/JOP gadgets, it's much better if the text section is only eXecutable.
(I'm sure there was another reason too, I'll add it if i remember)

Tariq

RogierBrussee, 2021-04-28:
All good points and if you want to actively discourage PC relative loads of constants in the text section, I guess one should not make those loads cheaper.

David-Horner, 2021-04-28:
There is a variant that is particularly relevant for RV64.as there is little value in RV32 unless F (not FinX) or D are included. 
That is, PC relative load and hop [skip or jump/branch] or alternatively called "immediate" loads.

The word or double word (or quad word for Q) following the instruction is loaded into the register and the next instruction resumes after the load value [effectively a branch over the immediate value].
This effectively puts a 64bit F load immediate, 96bit D [and RV64 X] load immediate and 160bit Q load immediate in the 32 bit address space. These are 48bit, 80bit and 144bit instructions respectively if the "load and hop" is in the 16bit opcode space.

tariqkurd-repo, 2021-04-29:
This intersting as it offers a different formulation of what I was considering.
We implemented: https://github.com/riscv/riscv-code-size-reduction/blob/master/existing_extensions/Huawei%20Custom%20Extension/riscv_LLI_extension.rst

as a 48-bit encoding, which has a 16-bit prefix and a 32-bit constant payload to load into a register.
The 48-bit encoding scheme is not ratified, so I haven't pursued this as part of Zce.

but if we formulate it as

<16-bit encoding> LKH/LKW/LKD
load-constant-half/word/double

LKH rd: rd = load[PC+2][15:0]; branch PC+4;
LKW rd: rd = load[PC+2][31:0]; branch PC+6;
LKD rd: rd = load[PC+2][63:0]; branch PC+10;

the load has special rules because it's instruction data, so can only cause I-side faults.
This is good for reducing code size and improving performance.

RogierBrussee, 2021-04-29:
One should be able to use the same mechanism to create long versions of JAL if one replaces the branch with small variations of JALR and JAL. 

An absolute version:

LJALABS32  rd:  rd = load[PC +2][31 :0];  (rd, PC) = (PC + 6, rd)  // Absolute,  effectively long version of JALR rd zero imm
LJALABS64 rd:  rd = load[PC +2][63:0];   (rd, PC) = (PC + 10, rd) 

Similarly using an hypothetical relative register version of JAL a relative version

LJAL32  rd:  rd = load[PC +2][31 :0];   (rd, PC) = (PC + 6,   PC + rd)   // Relative, effectively long version of JAL rd imm    
LJAL48  rd:  rd = load[PC +2][47:0];    (rd, PC) = (PC + 10, PC + rd) 

rd = zero should perhaps be disallowed, which should bot be  a problem because jumps of more than ±2^20 byte in a single subroutine should be rare (to say the least), and tail calls can use rd = t1 or an other non link register temporary.

David-Horner, 2021-04-29:
In general:

This "immediate op and hop" is applicable to more than just the load or 
JAL;

     any immediate op code could be expanded thusly, but diminishing 
returns for many ops with lower [static/dynamic] frequency and typical 
constant values that are within 12bit [+-2K] encoding..

tariqkurd-repo, 2021-04-29:
TBLJAL is the current proposed solution instead of LJAL

David-Horner, 2021-04-29:
LJAL and TBLJAL are complimentary.

TBLJAL has a limited number of vector elements and more memory footprint overhead for one-off targets.

LJAL has unlimited addressability and thus each invocation can target a different routine, collectively unlimited targets.
For one-off targets using LJAL:  the memory overhead is less than the TBLJAL [especially for RV64 when full range of addresses are considered and 32bit address/offset variants are included], in general is expected to be more performant [especially when gates are considered] and definitely lower system overhead [both in runtime setup/optimization for the limited entry slots and also csr maintenance].
Its simplicity will also be a factor in early acceptance.

I think there is room for competing solutions.
We should strive to make the alternatives non-adversarial and readily coexistent.
This likely requires some thinking outside the box, but is essential [to my way of thinking] to RISCV's outstanding success.
That is what I strive for, not just some adoption but pervasive, unfettered and aptly assisted adoption in the wide wild frontier.

tariqkurd-repo, 2021-04-29:
TBLJAL has significantly lower memory overhead for frequently called functions, and functions which are not frequently called do not contribute much to code-size.

The 32-bit encoding allows a table up to 8192 entries, although I may reduce this. We need to analyse it against the google v8 ELF file which has a 36MB text section, to see how many entries are worth keeping.

David-Horner, 2021-04-29:
>
> TBLJAL has significantly lower memory overhead for frequently called 
> functions,
agreed.
>
> and functions which are not frequently called do not contribute much 
> to code-size.
Of course you mean few static invocations vs dynamic call frequency.

But it is the number of entry points that count, not number of routines 
per se.

With high single use entry points [ definitely a design choice] LJAL 
would be the better apparatus when invocation is scattered throughout a 
large object file/load module.

We currently don't see much usage of this type, which may be due to less 
hardware [instruction] support for it than other alternatives.

> The 32-bit encoding allows a table up to 8192 entries,
although it may be reduced.
>
> We need to analyse it against the google v8 ELF file which has a 36MB 
> text section, to see how many entries are worth keeping.
A good empirical evidence example perhaps, but how indicative of the 
normal case?

As is mentioned above, what the current compilers generate can be far 
from optimal, even for such usage analysis.

Do we have good examples for the LJAL case?

Once again, I don't believe we are in a zero-sum game.

LJAL should at least be tabled. Others may have the time and incentive 
to analyze it on its merits and its application to specific use cases.

RogierBrussee, 2021-04-29:
FWIW, I agree with @David-Horner: LJAL  is a 48 bit JAL with a 32  bit immediate  (or 33 if you make the last bit implicitly 0). Hence, every AUIPC/JALR pair that cannot be converted to a TBLJAL or JAL is necessarily shorter as an LJAL. Therefore it should also greatly reduce the need for small CPU's to fuse those AUIPC/JALR pairs.

Thus counting the number of unconverted AUIPC/JALR pairs in a Linux kernel, Firefox, or a whole distro would also be test cases.

RogierBrussee, 2021-04-30:
I can think of two more reasons to seriously consider this version of C.LJAL.

1. It avoids an indirection so it is potentially more performant than TBLJAL in particular in the face of SPECTRE type mitigations  
2. An effectively 48 bit LJAL is cheap to do with C.LJAL prefix: 3 code points for LJAL with a link register  ra, t0 and t1 (or any other non link t register or even zero if that is considered not a problem) will already cover almost all use cases (although the prudent thing may be using 8 codepoints and a rd' = x0...x7 as link register). On the other hand, a 48 bit version requires a minimum  of 3* 31bit(or even 36 bit with a full 5 bit link register) worth of codepoints because the immediate (with last bit zero) becomes part of the instruction.  Since a 48 bit instruction comes with a 7 bit selector, that is a significant part of the instruction space.  

Mutatis mutandis, the same holds even more strongly for the long load immediate: to implement a 64 bit immediate like LDI or LFDI implemented as a 80 bit instruction with a 5 bit rd each requires 69 of the available 70 bit worth of codepoints.

tariqkurd-repo, 2021-04-30:
I've enquired about this type of encoding scheme, and been told to wait for 48-bit encodings. I did design an L.JAL instruction in the past (which is not very complex) which allows the full range of offsets. As and when 48-bit encodings are ratified I'm sure this, and the load 32-bit constant will be high up the list of new instructions.

closed 2021-04-30 by tariqkurd-repo

David-Horner, 2021-04-30:
Who are the nay-sayers?

Do we not get to face the competition?

I think that this is a big mistake in the context of support for all 
platforms.

We need a larger debate, and it is important that all the stakeholders 
are represented.

jhauser-us, 2021-05-02:
David Horner wrote:
> Who are the nay-sayers?

People with influence higher up in the approval process, with whom Tariq consulted.  Like it or not, there are people sitting on committees who have the collective authority to veto whatever a task group proposes, based on their own understanding of what would be best.  (And the assignment of this authority is not arbitrary, either.  There is a documented formal process by which the RISC-V Board of Directors is selected, and other committees below them.)

I don't have such a vote myself, but here is a snippet of what I wrote on the subject when asked:

> The big problem with extending a 16-bit instruction with an immediate constant is it effectively creates wider instructions without the same amount of thought about how easily these can be decoded by superscalar cores.  To help those superscalar cores, we really need first to develop a smart framework for instructions 48 bits and wider, *then* allocate instructions with large constants within that framework.

David-Horner, 2021-05-03:
> Who are the nay-sayers?
>
> People with influence higher up in the approval process, with whom
> Tariq consulted. Like it or not, there are people sitting on
> committees who have the collective [...]
When and where was this first determined?
>
> of what I wrote on the subject when asked:
When were you asked?
>
>     The big problem with extending a 16-bit instruction with an
>     immediate constant is it effectively creates wider instructions
>     without the same amount of thought about how easily these can be
>     decoded by superscalar cores.
Why the presupposition that insufficient thought was expended? Can this 
be rectified by expending more thought?
>
>     To help those superscalar cores, we really need first to develop a
>     smart framework for instructions 48 bits and wider, /then/
>     allocate instructions with large constants within that framework.
So how does this apply to embedded non-superscalar cores.

What of a smart framework that supports embedded and superscalar 
separately and distinctly.

This would help not only superscalars but all the ecosystem.

Making public these arguments and decision, especially for scrutiny and 
debate, helps everyone.
Do you have more, please!