riscvarchive/riscv-code-size-reduction #45: Tool chain:  CALL with alternate link register

RogierBrussee, 2021-04-27:
I noticed the following in the description of the push instruction

#64-bit AUIPC/JALR sequence
  e084be:	001f8317            auipc t1,0x1f8
  e084c2:	18a302e7            jalr  t0,394(t1) # 1000648 <__riscv_save_0>

I would suppose that comes from a toolchain.  I think this really ought to be

#64-bit AUIPC/JALR sequence
  e084be:	 ********            auipc t0,0x1f8
  e084c2:	 ********            jalr  t0,394(t0) # 1000648 <__riscv_save_0>

(note only t0's)  to allow the CPU to fuse this combination into an internal (or real)   
"L.JAL t0 ((0x1f8 << 12) + 1000648) " 
with a long immediate. The spec was changed with respect to how the CPU is supposed to handle return address stacks to allow this optimisation (see table 2.1 in section 2.5 of the spec).  It seems this is used for the corresponding case with ra (perhaps in the CALL macro) so not using it for t0 = x5  seems to be a mild pessimisation in the toolchain.

Sidenote: 
For the same reason I think that tail calls should not  be written

  e084be:	001f8317            auipc t1,0x1f8
  e084c2:	********              jalr     zero,400(t1) # 1000654 <__riscv_restore_0>

which can be fused in 

auipc      t1,0x1f8
"L.j"       ((0x1f8 << 12) + 400) # 1000654 <__riscv_restore_0>,

but like 

  e084be:	001f8317            auipc t1,0x1f8
  e084c2:	********              jalr     t1, 400(t1) # 1000654 <__riscv_restore_0>

which can be fused in 

"L.jalr"    t1, ((0x1f8 << 12) + 400) # 1000654 <__riscv_restore_0>

Note that t1 is NOT a link register, so it is harmless to set it while making the tail call and does not mess up the return address  stack tracking.  

However, this is probably more controversial and may reflect my ignorance of inner workings of the CPU.

tariqkurd-repo, 2021-04-29:
Hi @RogierBrussee , in all cases the core could fuse the two instructions - t1 is just a temporary value which is thown away. the Zce spec just reports what GCC10 generates.

David-Horner, 2021-04-29:
I have no difficulty in enhancing the definition of fuse-able [to be named fusible?] to include a temp register throwaways.

However, typically and classically fused refers to generating a result with no side effects as involving a throwaway register.
@RogierBrussee observation is valid, as is yours that GCC should generate fuse friendly code. There is no reason that it cannot. A peephole optimization will do, and it may already be on the books. As part of riscv-code-size-reduction it should at least be parked as an important objective to make gcc and LLVM friendly to fuse-able [vs fusible] constructs and by extension implementations. 

Emulation of some instructions require a temp register, e.g. BSET and family when rd and rs1 are the same.
To request the compiler to -ffusible(tempreg) render code that ensures the tempreg is available for throwaway over fusible instructions would allow implementations to "microcode with standard instructions" these operations, or a loader to expand to the corresponding sequence.

Both of these can be pursued by  riscv-code-size-reduction or at least parked as important observations/findings.

RogierBrussee, 2021-04-29:
No the core cannot fuse the two instructions precisely because t1 is _not_ thrown away, this is the whole point. It is of course the intention of the sequence that we don't care about t1 because t1 is caller saved, but the CPU does not know this and must dutifully execute the AUIPC instruction and set t1.  I already expected gcc to generate this sequence, but gcc should not, and does not if t1 is replaced by ra.  

I would assume that the loader is still instructed to replace the AUIPC t1 imm_high JALR t0 t1 imm_low.  combination with a JAL t0 (imm_low  + (imm_high << 12))  if the immediate fits as an optimisation.

tariqkurd-repo, 2021-04-29:
Hi @RogierBrussee you can file a GCC bug for this. Yes - it would be better if t1 were used throughout so only one register is written during the fused sequence.

tariqkurd-repo, 2021-05-10:
HI @RogierBrussee is it ok to close this now?

RogierBrussee, 2021-05-14:
I noticed on godbolt that both gcc and clang generate

call x5 register_save

so apparently it seems there is something wrong with the call macro. 

I tried to file a bug at gcc but couldn't register and gave up, also because I can't reproduce and only have the observation in your example.

I leave it to you what to do with  this issue. 

Rogier

tariqkurd-repo, 2021-06-24:
not for the code-size group to fix, sorry

closed 2021-06-24 by tariqkurd-repo