riscvarchive/riscv-code-size-reduction #46: some instruction clash with B-ext.

Xinlong-WU, 2021-05-05:
Hi all,
I found that there may some instruction clash, when I try to implement the Zce ext on LLVM.
I mean, some instructions have been defined by B-ext already, and they have different binary formate with them in B-ext.
for example:
`c.not` has been defined in B-ext in the following formate.
[image]

while in Zce proposal , it was defined as
[image]
and many other instructions have the same problem, like `c.neg`,`c.zext.w` etc.

does it is a mistake? or I miss something

Sincerely yours,
XinlongWu

jnk0le, 2021-05-05:
Those were removed from B to be handled by code size (RVA profile now gets `c.zext`/`c.sext` in Zcee). This table is the last thing remaining there.

tariqkurd-repo, 2021-05-05:
Hi @Xinlong-WU ,

yes - please ignore the encoding table in the B-extension.
I apologise but I am currently changing the encodings in Zce, because we had an encodings review and needed to make some changes.
I will release v0.4 (hopefully) next week - please update to that version when I release it. 
I will then try to minimise any changes.

closed 2021-05-07 by Xinlong-WU