riscvarchive/riscv-code-size-reduction #48: about usefulness of tbjalm

jnk0le, 2021-05-08:
The push/pop instructions are going to liberate any use of `-save-restore` milicode and they are still aggregated together with `c.tbl*`.

even in the provided example:
```
00e084be <vsprintf>:
  #64-bit AUIPC/JALR sequence
  e084be:	001f8317            auipc t1,0x1f8
  e084c2:	18a302e7            jalr  t0,394(t1) # 1000648 <__riscv_save_0>

00e084be <vsprintf>:
  e084be:	<16-bit>            tbljalm #x ;#<mapped to __riscv_save_0>, saving 6-bytes
```
that's definitely a place for `c.push`.

yet, `c.tbl*` instruction is divided into; 8 points for `c.tblj` and 56 points for `c.tbljalm`.

Shouldn'tthose be in the reverse order? The `c.tblj` falls into popular tail-call pattern and the latter one sounds like a false positive from benchmarking (like many milicode islands for proximity, etc).

Is there any other popular use of milicode I'm not aware of?

abukharmeh, 2021-05-09:
Hello jnk0le, 

I think you are correct, I think that's a typo in the spec, they are reversed! 

Kind regards,
Ibrahim.

tariqkurd-repo, 2021-05-10:
Yes they were, please check Zce v0.41

abukharmeh, 2021-05-14:
@tariqkurd-repo  This is still not completely fixed, under the table C.TBLJALM and C.TBLJ linked registers are still swapped !

tariqkurd-repo, 2021-05-17:
fixed!