riscvarchive/riscv-code-size-reduction #49: MULI is overlapping with SLLI

Lucas-Wye, 2021-05-10:
imm[11:0] | rs1 | 001 | rd | 0010011 | MULI
-- | -- | -- | -- | -- | --

0000000 | imm[4:0] | rs1 | 001 | rd | 0011011 | SLLI
-- | -- | -- | -- | -- | -- | --

tovine, 2021-05-10:
Opcode is different
[image]

Lucas-Wye, 2021-05-10:
I copied the wrong opcode before, here is the correct one.

| 0000000 |  shamt | rs1 | 001 | rd | 0010011 | SLLI
| --| --| --| --| --| -- | --

tariqkurd-repo, 2021-05-10:
ok - please see Zce_spec 0.41. I've moved MULI into custom-0. We will have to find a new encoding in the future, but it's ok for the prototype implementation.

tovine, 2021-05-10:
Isn't there anywhere in the existing M extension opcode space we could put it? I kinda feel it belongs there... 😄
For example take one of the remaining funct3 values in the RV64M opcode (`0111011`)?
Looks like both `001`, `010` and `011` are available there.

tovine, 2021-05-10:
And on that same note: do we need `MULIW` (for RV64) and `MULIH` as well?

tariqkurd-repo, 2021-05-10:
If I put in the RV64 section I'll be unpopular and it will have to move again.
For the prototype this is good enough for now (the TBLJAL CSRs don't have real addresses either yet).

tariqkurd-repo, 2021-05-11:
@Lucas-Wye can you close this issue now please? Thanks!