riscvarchive/riscv-code-size-reduction #53: areg moves in push are also unsafe

jnk0le, 2021-05-18:
POP instructions already lost s0 to a0 move as interrupted sequence will load to s0 and then redo move.

PUSH can also do that with areg moves:

`c.push  {ra, s0-s5},{a0-a3}, -64`

push ra, s0-s5
move a0-a1 into s0-s1
INTERRUPT
and then push any overwritten s register on stack over their original content

Instead of removing also areg moves, I would recommend to add s0 to a0 back for pop instructions (if that has benefit) and let the microarch to choose between:

- whole sequence uninterruptible (can write sp at the beginning if that simplifies things). **Adds jitter to isr latency
- interruptible only in "safe" areas of the sequence  **less jitter that is less consistent/verifiable for worst case isr latency scenario
- extra CSR state for resumable push/pop like in the "other architecture" **must be in existing csr that's already stacked in handlers (incl nesting), to be "microarch choice" only. Otherwise it's disruptive arch addition.

The 3rd one as disruptive addition can still be microarchitectually "degraded" into the former two, to save on transistors.

I see that someone is already complaining about modified state after partial execution.
https://lists.riscv.org/g/tech-code-size/topic/push_pop_restarting/82620542?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,82620542

jnk0le, 2021-05-18:
PUSH in table 40 is also missing `areg_list`

tariqkurd-repo, 2021-05-19:
the PUSH encoding doesn't include areg_list- it implicitly moves 

a0 into s0 if s0 is saved
...
a3 into s3 if s3 is saved

and so saves encoding space.

for interruptibility - after the final load or store interrupts are not permitted, so the sequence is safe.
I've added a comment into the pseudo-code to show when interrupts are no longer permitted.

I won't extra extra-state for restarting the sequence - it adds a lot of complexity - and was very unpopular when originally proposed. The micro-arch can always choose not to allow interrupts but it can't really choose not to allow watchpoint triggers until after the sequence completes, so there will always be some kind of interruption possible.

jnk0le, 2021-05-19:
Now I see the option 2 is current PoR, due to busfaults.

Without returnable CSR state CLIC will have to add another jitter cycles to interrupt latency.
Also some microarchs are not able to execute moves in parallel with stores 

I have opened separate issues about side topic (#56,#57) so closing this if csr state was already rejected.

closed 2021-05-19 by jnk0le

tariqkurd-repo, 2021-05-20:
>Also some microarchs would not be able to execute moves in parallel with stores

yes - this is true - to ensure that the sequence is restartable