riscvarchive/riscv-code-size-reduction #54: complexity of sp offset calculation in push/pop instructions

jnk0le, 2021-05-18:
The base for load/stores in push/pop instructions is currently calculated by the following formula:

`sp +/- (spimm + space resulting from reglist)`
it looks like 3 source addition of:
- register
- constant
- sort-of-constant

It is imaginable that in case of compressed push/pop, the addition of spimm and reglist space can happen in 16 to 32bit decoding as (and especially) those constants are quite small.
But the 32 bit versions also have this feature, so all constant additions have to happen in later stages anyway.

I'm not quite sure how it translates into real designs but maybe we should leave 32bit encodings with absolute offsets only.
Compilers can error in case of too small offsets. Throwing illegal instruction exception in those cases shouldn't be a big hassle.

tariqkurd-repo, 2021-05-19:
Good question, thanks for reviewing the spec so carefully.

For C.PUSH it's fine, it's always SP-4 for the first step.

For C.POP/C.POPRET it's more complex - but not much more - it can just be a lookup table. I haven't enumerated the cases in the spec but it's just a constant offset from SP, and the constant will be output from the decoder as it's a simple function of rlist and spimm. To answer the question a different way, the adder for the two constants will be implemented in the decode phase (if it is implemented as an adder and not just a LUT), and if there's extreme timing pressure could be in parallel with the register file read. I suspect the area will be similar between LUT and adder, and the LUT would be faster.

Therefore the first step in the sequence will simply execute as ld ra, offset(sp)
with no additional add needed in the execution.

tariqkurd-repo, 2021-05-24:
@jnk0le can you close this now or do you have further concerns?

closed 2021-06-01 by tariqkurd-repo

jnk0le, 2021-06-07:
>For C.PUSH it's fine, it's always SP-4 for the first step.

`sp-4`? I think it's `sp - (spimm + space resulting from reglist) + 0`
```
sw (-(spimm + space resulting from reglist)+0)(sp)
sw (-(spimm + space resulting from reglist)+4)((sp)
sw (-(spimm + space resulting from reglist)+8)((sp) 
[...]
add sp, sp, (-(spimm + space resulting from reglist))
```
Well, it looks like 3 constants + register.

Due to lack of returnable state sp cannot even be written immediately and fire sequence of `sw 0(sp), sw 4(sp) , sw 8(sp) ...`, there needs to be some kind of temporary register in sequencer. First instruction becomes more likely to not be a store.

In case of pop/popret it's simple sequence of  `lw 0(sp), 4(sp), 8(sp) ...` and then write to sp, calculated offset.

>and if there's extreme timing pressure could be in parallel with the register file read. I suspect the area will be similar between LUT and adder, and the LUT would be faster

for 16 or 32 bit versions? 

LUTs/adders should be obviously much less complex in case of 16bit. Currently it has to be expanded to 32bit instructions as is and then decoded by LUTs/adders.

jnk0le, 2021-06-07:
ok, I was wrong. Registers are stacked from top to bottom. So the pop/popret are the complex cases.