riscvarchive/riscv-code-size-reduction #57: 32bit PUSH no longer has control over areg_list

jnk0le, 2021-05-19:
Looking at table 40, push instruction has a hole in place that pop/popret has ret_val field.

Reading the spec gives impression that it used to have some mv control:

>Increase PUSH/C.PUSH embedded moves to include a3 [...] PUSH has the option not to do any so as not to waste performance

>Additional benefits of the 32-bit encoding are:
>disabling the mebedded mv instrucitons for PUSH if they are not needed

>push {ra,s0-s2}, -64 must use a 32-bit encoding as {a0-a2} is excluded

>#-O1/-O2/-O3: compile to PUSH to avoid the redundant register moves of a2, a3 even though the code is larger

and especially table 54 that remained four `mv` values (unknown purpose)

I think we could add back `mv` control field in that hole. 4 values allows to specify none and 3 different ranges (for performance reason as is the case for 32bit encoding)

tariqkurd-repo, 2021-05-19:
yes - there was redundant text there. We decided to remove the programmability to simplify the instruction spec, otherwise it covers a really large number of cases. If there's a performance concern then use the 32-bit PUSH and exclude `areg_list`, otherwise in general you can use C.PUSH and include the moves.

Is that clearer now?

jnk0le, 2021-05-19:
>If there's a performance concern then use the 32-bit PUSH and exclude areg_list

The problem is that 32bit PUSH doesn't have a way to disable areg moves

tariqkurd-repo, 2021-05-19:
You're right!! I'll fix it.

tariqkurd-repo, 2021-05-19:
Is it ok now?

closed 2021-05-19 by jnk0le