riscvarchive/riscv-code-size-reduction #58: Code savings on 'summary' tabs do not match code savings from 'stats for Zce 0.41' tab

Silabs-ArjanB, 2021-05-20:
The 'total saving' numbers on the 'summary' tabs of https://docs.google.com/spreadsheets/d/1UYll7HGR_QLGTsHcjGoNL4EodM5BNO41hXdxVAFaxFs/edit#gid=1281210325 do not seem to match weighted average of the numbers in the 'stats for Zce 0.41' tab.

It seems to me that the spreadsheet computation does not understand the 'C.TBLJ*' syntax that is used on the summary slides. The 'wildcard' is not taken into account as far as I can see.

tariqkurd-repo, 2021-05-20:
Thanks @Silabs-ArjanB I've fixed and and found some other issues too, all the LLVM results are LLVM13, nothing is from LLVM13 any more. Is it ok now?

Silabs-ArjanB, 2021-05-20:
I double checked some other numbers and it seems that the weighted average 'PUSHPOP wrt MSR' score, i.e. AA112 on tab 'stats for Zce 0.41' is below each of the values in AA53-AA56, which does not seem correct.

abukharmeh, 2021-05-20:
Thank you, I have just fixed that one as well.

closed 2021-05-21 by Silabs-ArjanB