riscvarchive/riscv-code-size-reduction #59: Feedback on Zce extension proposal v0.41.2

Silabs-ArjanB, 2021-05-21:
Hi,

Here is some (mostly minor) feedback and questions on the Zce extension proposal v0.41.2.

- Chapter 3

    BEQI, BNEI from section 6.5 are missing from the figure and tables of chapter 3.
    C.TBLJ, C.TBLJALM from section 6.6 are missing from the figure in chapter 3
    DECBNEZ (32-bit) from section 7.1.2 is missing from the figure and tables of chapter 3.

    > Zcee is a simpler subset of Zcea for high performance cores, it can be implemented without the test of Zcwa

    What is Zcwa?

- Chapter 4

    Typo: Zc321 -> Zce32

- Section 8.5.2

    What does MADDR mean? Should it be MTVAL?

- Section 8.5.4 

    Possible table caching is mandated to be flushed upon a write to xTBLJALVEC. I understand/agree that the cache could get stale, but I wonder why a hardware mechanism is proposed for flush/refetch. Of course the actual cached table content could also get stale by execution of regular store instructions to the vector table. The regular way to deal which such store vs. instruction fetch interaction would be by using fence.i instructions. Note that RISC-V already supports the MTVEC CSR and an implementation could also decide to cache that related vector table. There is no hardware requirement to flush anything when the MTVEC CSR is written, so my question is why does xTBLJALVEC need to be treated differently? Could a sequence using fence.i be recommended instead?

- Section 9.5.1

  > if (misa.MXL==1 && sp[2:0]) {take_illegal_instruction_exception();}

    I understand that there is a stack alignment restriction. My question is why the stack pointer is being checked in hardware (possibly leading to an illegal instruction exception) whereas the original sequence that is being replaced by a push or pop did not implement that check (and could never result in an illegal instruction exception)? Stack checking is nice of course, but there are many more checks we could do on the stack pointer and doing them only for push/pop seems a little random. Shouldn't SP checking be an extension in itself (outside of Zce)?

- Section 9.8

    Can the impact on MTVAL be described for load access faults and store access faults (e.g. should it contain the address of the faulting load/store address, and if so, the first faulting address or any faulting address?) and for any other of the reasons that could possibly update MTVAL?

- Section 9.5.2
    
    Why is the following part of the uninterruptible code; wouldn't it be okay if this was executed multiple times?
    
        > 
        > if (ret_val) {
        >    switch(ret_val) {
        >       "0":  asm("li a0, 0");
        >       "1":  asm("li a0, 1");
        >       "-1": asm("li a0, -1");
        >    }
        > }
    
    Also note that the pseudo code from 9.5.2 and 9.9.2 have the LI and ADDI in different orders (both are okay I believe)

- Section 9.9.3

    The following seems expensive to implement or even conflicting to me:

    It is required to:
    
    > Detect any virtual memory page faults or PMP faults for the whole instruction before starting execution (instead of during the sequence)
    
    It is possible that the following will occur during execution:
    
    > Load/store access fault

    A PUSH/POP based store/load sequence can easily cross PMP regions and therefore it is expensive to check the whole instruction before starting execution. As the spec allows load/store access faults during execution, why can't the PMP fault not be checked _during_ the instruction as well? (Of course  there would be no way to safely resume such faulted instruction). Or another way of asking the question. What would cause the allowed load/store access faults and why the same faults not allowed to be cause by the PMP?

- Section 9.12
    
    The replacement sequence would only be safe if it cannot get interrupted after the ADDI (otherwise the save sequence upon interrupt entry would push content relative to a wrong SP).

- Section 11.4.3

    > The core to jump to address m

    The core actually jumps to (a slightly modified) version of the memory content at address m.

tariqkurd-repo, 2021-05-24:
Hi @Silabs-ArjanB , thanks for great feedback!

I've fixed the trivial ones.

8.5.4: Table caching - fence.i would be the regular way to solve this, I agree.  This adds the requirement that Zifencei is included to allow caching, which is reasonable.

9.5.1: Stack alignment check: this is present because the floating point versions PUSH.F etc. need the stack to be aligned for the sequence to work when moving from X to F registers, if XLEN != FLEN, so it's to keep the definition regular between them. Given that I have postponed PUSH.F etc. maybe we don't need this check. It seems to add an exception source from (probably) the ALU which is irregular in RISCV.

9.8: I've extended the text, please review it.

9.5.2: yes, the "li a0, xx" can be executed more than once

9.8 (not 9.9.3)

tariqkurd-repo, 2021-05-25:
Hi @Silabs-ArjanB , with the possible exception of 9.5.1 are you happy with the fixes to the spec?

Silabs-ArjanB, 2021-05-25:
Hello @tariqkurd-repo,

Thank you for your quick follow up. 

Here is the remaining feedback:

- The figure from chapter 3 is missing the C.TBLJ, C.TBLJALM and DECBNEZ (32-bit).
- With respect to 9.5.1 I was mainly triggered by the fact that no such hardware check was required before. Thank you for the added explanation with respect to PUSH.F. I would probably prefer to not have the stated SP check in hardware for the current PUSH/POP definition, but I am okay if you want to be 'forward compatible' with a possible push.f.
- Thank you for updating 9.8. I am fine with the text, but I expect that you mean MTVAL when you write MADDR?
- My reference to 9.9.3 was wrong indeed. I actually meant 'item 3 of section 9.9.5'. I agree that in 9.8 it is an implementation choice to perform PMP/VN checking before or during execution. For '9.9.5 item 3' I however still do not understand why that extra requirement is added (as it will be expensive to implement). Why can't we again leave it to the implementation to either check before or during execution? Of course if the check fails during execution for non-idempotent memory, then there is no good way to recover from it, but that is probably the case as well if the error would be guaranteed to happen 'before' execution.
- The text of 11.4.3 seems not fixed yet (in v0.41.3). It still implies that there is a jump into the table as opposed to a jump based on the content of the table.

tariqkurd-repo, 2021-05-25:
OK, please review it again

Silabs-ArjanB, 2021-05-25:
Looks good; thank you!

closed 2021-05-25 by Silabs-ArjanB