riscvarchive/riscv-code-size-reduction #61: decoding conflict of C.DECBNEZ

Marcythm, 2021-05-29:
```
Decoding Conflict:
                101_..........10
                101...........10
                101.............
                ................
        C_DECBNEZ 101___________10
        C_FSDSP 101___________10
```

Marcythm, 2021-05-29:
oh sorry, I haven't noticed that the `c.fsdsp` instruction requires D-extension

Marcythm, 2021-05-29:
I found a new decoding conflict: the encoding of `c.lb` (in spec 7.3) conflicts with `c.decbnez` (in spec 7.1.1)
an example: the encoding of `c.decbnez s0, 1, -4` and `c.lb s0, 2(s0)` are both `[0x22, 0xa0]`

tariqkurd-repo, 2021-05-29:
Table 30 was incorrect, it had a mistake in the encoding of C.LB, C.LH - they shoudl be in the 001____10 encoding group not 101____10. Is it ok now?

Marcythm, 2021-05-29:
ok, thanks

closed 2021-05-29 by Marcythm