riscvarchive/riscv-code-size-reduction #63: Could you provide  test cases for the gcc ?

LiaoShiHua, 2021-05-31:
I added some builtin templates to gcc, and I want to get some testcases to test them。

David-Horner, 2021-05-31:
> I added some builtin templates to gcc, and I want to get some testcases to
> test them。
Please describe these templates or direct us to your changes.

Thanks.

LiaoShiHua, 2021-06-01:
Most of them come from simple modifications of similar instructions in other processors, I don’t know if this is correct。  
So，I want to get some testcases to test them。  
https://github.com/LiaoShiHua/riscv-gcc/blob/riscv-gcc-zce/gcc/config/riscv/zce.md

tariqkurd-repo, 2021-06-01:
Hi @LiaoShiHua I'm not famliar with the syntax in that file. Note that some of the instructions listed have been removed from the spec already e.g. ADDIADDW.

PLCT are developing the simulators (Spike/QEmu) and the toolchains (GCC/LLVM). It is difficult to develop any testcases without a simulator and a toolchain (at least an assembler), so maybe someone in PLCT can develop some simple testcases?

closed 2021-06-01 by LiaoShiHua