riscvarchive/riscv-code-size-reduction #64: the syntax of C.MVA01S07

Marcythm, 2021-06-01:
the example syntax of C.MVA01S07 given in table 16 is `mv (a0, a1), (sreg1, sreg2)`, but in table 15 I also see the usage `C.MVA01S07 sreg1, sreg2` (though it is not given as a syntax). Which is the correct syntax?

tariqkurd-repo, 2021-06-01:
table 15 was wrong, the correct syntax is `mv (a0, a1), (sreg1, sreg2)`

Marcythm, 2021-06-04:
Here I have a question: since the two operands `a0` and `a1` are fixed, why the syntax of `C.MVA01S07` was designed to be `mv (a0, a1), (s0, s1)` but not `c.mva01s07 s0, s1` ?

tariqkurd-repo, 2021-06-04:
good question -  it's because I also deigned a generic mv instruction, and I wanted the syntax to fit.
but you're right, it should be `c.mva01s07 s0, s1`.
do you want me to change it?

Marcythm, 2021-06-04:
yes of course, that will greatly reduce our workload while having little impact on the overall instruction set design. thanks!

tariqkurd-repo, 2021-06-04:
that's great - it's better this way.

tariqkurd-repo, 2022-02-08:
let's go with the current spec, c.mva01s sx, sy

closed 2022-02-08 by tariqkurd-repo