riscvarchive/riscv-code-size-reduction #66: [c.]push.e/[c.]pop.e/[c.]popret.e are designed for RV32E only

jnk0le, 2021-06-07:
Even though [eabi proposal](https://github.com/riscv/riscv-eabi-spec/blob/master/EABI.adoc) doesn't forbid RV32I or RV64I, the Zce pushpop spec assumes that the only valid target for eabi is RV32E. 

16bit encoding has rlist3 crippled to rlist2 to not cover anything above `s4`.
32bit version is using previously reserved upper 3 rlist points to not cover anything above `s4`.

Worser thing is that EABI is going to have even more saved registers so we may se gimmicks with UABI push/pop used in EABI programs (if UABI and EABI versions of those instructions will be available at the same time).

Maybe there was not enough compressed space but 32bit can definitely use 1 extra bit from funct9 or field2 to do `.E` selection 
(overlap of  first first 3 points can be used to cover additional number of saved registers)

tariqkurd-repo, 2021-06-08:
Do you have a genuine application for the EABI on an architecture other than RV32E?

jnk0le, 2021-06-09:
>The EABI calling convention is designed to work the same on all RISC-V embedded targets with the same XLEN (i.e., for 32-bit systems, RV32I, RV32E, with or without the Zfinx option).

this text desn't forbid RV64 EABI.

there is also a reference to 64bit
>The stack alignment for XLEN=64 systems is 16 bytes.

>Do you have a genuine application for the EABI on an architecture other than RV32E?
Maybe 

EABI is intended to reduce interrupt latency.  Due to reduced number of caller saved registers, there will be even more pressure on cellee saved registers, so the:
>c.push.e etc. already cover the most common cases - it is unusual to push/pop more than a few saved registers

might not be the case anymore.

RV32E gets "normalized" register layout allocation, though I'm not sure how it affects code density/performance.

>We can only allocate 16-bit encodings if they will make a big difference to code-size.

Apart from irregularity of `spimm0_5` and `ret0` encoding, I'm not advocating to expand those. My point is that the 32bit encoding has definitely enough space to add `.E` selector and leave upper rlist points reserved for special cases.

tariqkurd-repo, 2021-06-09:
RV64 EABI isn't forbidden but there are no plans to implement it.
RV64E also isn't forbidden but there are no plans to implement that either, if it is implemented it will need a version of the EABI.

For the case of RV32I + EABI - there are a lot of saved registers - and it's really not clear how many will be used. Allowing PUSH/POP to flexibly save up to 22 registers is regular but extends the hardware state machine, adds more verification cases and so increases the complexity and workload. Therefore this can only happen once we have the EABI proposal frozen (which also affects C.PUSH.E etc. as s4 may not be the highest numbered s reg in RV32E) and we can also try compiling code to see the distribution of saved registers.
I'll add this as a future idea.

tariqkurd-repo, 2021-06-22:
let me know if this needs reopening

closed 2021-06-22 by tariqkurd-repo