riscvarchive/riscv-code-size-reduction #69: Tail calls and tbljalm/tblj/tbljal

RogierBrussee, 2021-06-13:
There are two ways to call the same function: 

jal ra func1        (linked call)
and
j func1               (tail call*)

or for milicode

jal t0 func2.     (linked call)
and
j func2             (tail call)

with a similar distinction for an AUIPC/JALR pair ((CALL ra func1 vs TAIL func1 or CALL t0 func2 vs TAIL func2). 

The point is that what the link register of a function is, is part of the "ABI"  of the function, while whether to call it linked or jumped to as a tail call is not. This means that it is likely that the majority of the  functions in the tblj table will also be in the tbljal tables. This then means that a different design that uses one bit in the instruction to distinguish tail call or linked call may be reasonable.  Whether the entry has link register t0 or ra can still be determined from the index part of the instruction e.g. the first 8 entries with link register t0  and 120 entries with link register ra (or is only to be used as tail call or is otherwise only jumped to).  

The tradeoff here is of course that it can give smaller tables than the current design and fewer cache entries because there are no duplicates, but that for the same number of bits in the C.TBLJ[AL[M]] instruction the index has to become 7 bits and there can be only 128 entries. 

It is not clear to me what the best tradeoff is, and maybe this tradeoff should be mentioned in the documentation.

tariqkurd-repo, 2021-06-14:
Hi @RogierBrussee I suspect that this is a rare case. I'll ask my colleague @abukharmeh to check for you.

RogierBrussee, 2021-06-16:
Thanks! Nothing like real data to settle this!

abukharmeh, 2021-06-18:
Hi, I checked this for our normal benchmarking suite, and its indeed there, but I am not sure its significant enough to justify changes in encoding space ?! 
```
File name, Number of duplicates
GCC10_audiocodec_fixed_LC3plus.elf 2
GCC10_audiocodec_fixed_opus_demo.elf 1
GCC10_audiocodec_float_LC3plus.elf 0
GCC10_audiocodec_float_opus_demo.elf 1
GCC10_coremark.elf 0
GCC10_embench_aha-mont64.elf 0
GCC10_embench_crc32.elf 0
GCC10_embench_cubic.elf 0
GCC10_embench_edn.elf 0
GCC10_embench_huffbench.elf 0
GCC10_embench_matmult-int.elf 0
GCC10_embench_minver.elf 0
GCC10_embench_nbody.elf 0
GCC10_embench_nettle-aes.elf 0
GCC10_embench_nettle-sha256.elf 0
GCC10_embench_nsichneu.elf 0
GCC10_embench_picojpeg.elf 0
GCC10_embench_qrduino.elf 0
GCC10_embench_sglib-combined.elf 0
GCC10_embench_slre.elf 0
GCC10_embench_statemate.elf 0
GCC10_embench_st.elf 0
GCC10_embench_ud.elf 0
GCC10_embench_wikisort.elf 0
GCC10_fpmark_atan-1M.elf 1
GCC10_fpmark_inner-product-mid-10k.elf 1
GCC10_fpmark_radix2-sml-2k.elf 1
GCC10_fpmark_xp1px-sml-c100n20.elf 2
GCC10_iot_huawei_application.elf 18
GCC10_iot_huawei_protocol.elf 21
GCC10_iot_zephyr_central.elf 2
GCC10_iot_zephyr_peripheral.elf 6
GCC10_testsoftfloat.elf 0

```

tariqkurd-repo, 2021-06-18:
Thanks @abukharmeh so we see most cases there are very few, but in the real Huawei IoT code we do have ~ 20 duplicates. This covers about 1/3 of the TBLJ entries, but only 10% of the TBLJAL entries.
I've thought about whether we could have some overlap between the tables but I can't think of any solution which is more efficient than the current one. I think the amount of duplication is small enough not to change the current scheme.

What do you think @RogierBrussee ?

RogierBrussee, 2021-06-18:
Thanks  @abukharmeh and @tariqkurd for taking this serious!

I don't know how large the Huawei IOT code is but it seems that the larger the code the more likely it is that a function worth putting in as a TBLJ entry also enters as a TBJAL entry. So maybe the Linux kernel would be an interesting testcase. 

In any case I did think of an encoding that gives "shared" table entries but does not reduce the number of available entries quite as much.  We can use the 8bits worth of entries like

00000imm == 0..15         map to jal t0  entry[00000imm]
0imm          == 16..127     map to jal ra  entry[0imm]               (imm >= 16)
10000imm == 128..135   map to  jal t0 entry[10000imm]
10imm.       == 136..191   map to  jal ra  entry[10imm]             (imm >=16)
11imm        == 192..255   map to jal zero entry[10imm]           #note 10imm!!!

That way there are 128 + 64  = 192 entries in total
16 tbljalm entries
  8  entries that can be used as tbljalm or tblj depending on how they are called in the code stream
112 tbljal entries 
  56 entries that can used as tbljal or tblj depending on how they are called in the code stream. 

I don't know if it is worth it, in particularly what it does for caching the call/jump targets, but I thought I mention it anyway. 

Rogier

tariqkurd-repo, 2021-06-21:
Sorry @RogierBrussee I don't understand your table. What do "0000imm" "10imm" etc. mean? I think you are specifying bits of the imemdiate but can you list them more clearly.

@abukharmeh how about running the analysis on v8 / Debian? To see if we get different results?

RogierBrussee, 2021-06-21:
Indeed I tried to specify the 8 bit of the  immediates of TBLJ*. Just enumerating them (with a space separator after 4 bits purely to avoid me miscounting zeros and ones.) gives (this is a bit different from what I tried to express above: while doing the enumeration I realised that the original coding reserved 16 rather than 8  of 64 entries for shared tbljalm/tblj)

TBLJALM:

0000 0000    -->   jal t0 entry[0000 0000]
...
0000 0111.    -->   jal t0 entry[0000 0111]

0100 0000    -->   jal t0 entry[0100 0000]
..
0100 0111     -->   jal t0 entry[0100  0111]

1000 0000  -->    jal t0 entry[1000 0000]     #NOTE Shared entry with TBLJ    
...
1000 0111   -->     jal t0 entry[1000 0111]      #NOTE Shared entry with TBLJ
----------------------------------------------

TBLJAL

0000 1000.  -->    jal ra entry[0000 1000]
...
0011 1111     -->.    jal ra  entry[0011 1111]

0100 1000   -->    jal ra entry[0100. 1000]
..
0111   1111.   -->.   jal ra entry[0111   1111]

1000 1000  -->     jal ra entry[1000 1000].  #NOTE Shared entry with TBLJ
...
1011 1111    -->      jal ra entry[1011 1111]      #NOTE Shared entry with TBLJ

---------------------------------------------------------------------------------
TBLJ

1100 0000 -->      jal zero entry[1000 0000].  #NOTE Shared entry with TBJALM 
...
1100 0111. -->       jal zero entry[1000 0111]    #NOTE  Shared entry with TBJALM
1100 1000  -->      jal zero entry[1000 1000].  #NOTE  Shared entry with TBLJAL
....
1111 1111    -->       jal zero entry[1011 1111].     #NOTE Shared entry  with TBLJAL

Rather than sacrificing a full bit for choosing between the link register and zero  the coding only does this for half the range effectively sacrificing half a bit. It also,  tries to make recognising when to use t0, when to use ra and when to use zero easy

if(11** ****)
   use  zero
else if(**00 0***
   use t0
else
  use ra

Other codings implementing similar ideas (or different numbers of entries for TBJALM)  are of course possible.

tariqkurd-repo, 2021-06-21:
ok - so you are overlaying table entries onto the same address, which effectively reduces the size of the table.
for example one table entry maps to 1100 0000 TBLJAL also maps to 1000 0000 TBLJALM
So you are effectively sacrificing entries in the encoding to save memory space. I would say that memory space is cheaper than encoding space.

It also adds constraints on the table - I've already divided it up between the three encodings, but it puts more constraints on it.

Let's see what results @abukharmeh comes up with.

Tariq

RogierBrussee, 2021-06-21:
Yes it reduces the size of the table that is why it is a tradeoff as well, just a different tradeoff (but it has overlaps between TBLJAL and TBLJ and TBLJALM and TBLJ not between TBLJALM and TBLJAL).

I agree that memory is cheaper than encoding space, so I think the real question is what this encoding does to caching: if I understood correctly, while the 256 or 192 encodings are indices in tables of function pointers in memory, you can also use them as  keys to function addresses in some processor internal cache, and this cache is per process/ addresspace identifier. So if you knew that _all_ of the TBLJ entries would also be TBLJAL or TBJALM entries then this proposal would have the same number of function entries as the original one (192), but save cache-keys by not having different keys for the same functions.  The smaller tables in memory is an added bonus.  Of course that is the extreme situation.  

Again I do not know if it is worth it or not, but the data from @abukharmeh should give the info for an informed decision.

yours
Rogier

tariqkurd-repo, 2021-06-21:
>and this cache is per process/ addresspace identifier

the main use for table jump is for small embedded cores, with one static table setup by the linker for all applications to use.
In terms of caching, I don't expect the whole table to be cached, the just the few most frequently accessed of each category. I agree though that it would be unfortunate to cache the same address more than once - but this would only happen if the same function was called frequently both without and without linking.

The only overlap we have so far is TBLJAL/TBLJ calling the same function. TBLJALM has no overlaps, as these currently  refer to the -msave-restore functions, and I'm not aware of any other milicode in use in the compiler.

So for the argument to be compelling I think that both the TBLJAL/TBLJ call to the same function would both need to be high frequency calls. We're only doing static analysis - we don't execute the benchmarks - so we only see how many times each is called in the ELF.

tariqkurd-repo, 2021-06-24:
we'll look at this when we have a real table jump implementation and some performance analysis, this will be in the autumn

tariqkurd-repo, 2021-10-19:
there's some overlap but not enough to change the spec.

closed 2021-10-19 by tariqkurd-repo