riscvarchive/riscv-code-size-reduction #71: encoding formate of c.popret

Xinlong-WU, 2021-06-15:
in spec, we find statements that
`c.popret {ra,s0-s2}, {0}, 32 can use a 16-bit encoding`
and 
`popret {ra,s0-s2}, {1}, 32 must use a 32-bit encoding as c.popret only supports 0 as the ret_val value`
Dose it means the value of `ret0` of c.popret will be 0 forever. Can I write it as a static number?

tariqkurd-repo, 2021-06-15:
hi - no -it's an option
`c.popret {ra,s0-s2}, {0}, 32` and
`c.popret {ra,s0-s2}, 32`

are both valid

but if you want to return 1 instead you must use the 32-bit encoding

`popret {ra,s0-s2}, {1}, 32`

sometimes function return values are not 0 or 1, other values are also possible and we can't include all possible values in the `popret` instruction.

is that clear?

Xinlong-WU, 2021-06-15:
ok， so does `popret {ra,s0-s2},  32` is still valid?, cause we all know we can use `popret` instead of `c.popret` only when return 1.

tariqkurd-repo, 2021-06-15:
`popret {ra,s0-s2}, 32` is valid.
You choose whether to include a return value.
For c.popret the return value is 0, or no return value
For popret the return value is -1, 0, 1 or no return value

tariqkurd-repo, 2021-06-17:
Hi @Xinlong-WU , is it clear now? Can you close this issue, or do you need me to fix the spec?

tariqkurd-repo, 2021-06-17:
In fact I have just updated the syntax, so that the return value must always be specified, for example

`popret {ra,s0-s2}, {}, 32` for no return value, or
`popret {ra,s0-s2}, {1}, 32` to return 1

is that ok for you?

Xinlong-WU, 2021-06-19:
that's good. Thanks.

closed 2021-06-19 by Xinlong-WU