riscvarchive/riscv-code-size-reduction #74: The spimm of C.POP[.E]

Xinlong-WU, 2021-06-19:
it uses spimm[4] for `c.pop`, only one bit.
and considering `spimm[4]` is `1` if `spimm == 16`

does it means the stack  adjustment is 16  if `spimm[4]` is 1
and it is 32 if `spimm[4]` is 0

in other words, whether only 16 and 32 are valid for C.POP[.E]

tariqkurd-repo, 2021-06-20:
Hi - Table 51 has the answer - it shows the spimm range.
It depends on the number of registers - spimm[4]=0 has the minimum stack adjustment (16 or 32) and spimm[4]=1 adds 16 so 32 or 48.
I've found a mistake in table 51 - which I will fix now....

Xinlong-WU, 2021-06-20:
yes, but table 51 just point out the range like `16 to 32`. it confuse me a lot.

cause we just pick one bit of spimm. it means some issue will happen when disassemble.

eg.

| number | binary|
|----|----|
|31| 0001 1111 |
|30| 0001 1111 |

they have same number in 4th bit.

i don't know how to handle it when I  disassemble the code.

tariqkurd-repo, 2021-06-21:
HI @Xinlong-WU - spimm is always 16-byte aligned - so spimm[3:0]=0.
Therefore I only specify spimm[6:4].
In table 51 the minimum stack adjustment is always with spimm=0. The minimum value is the memory required for the registers, rounded up to 16-byte blocks. The extra range is from setting a non-zero value in spimm.

If table 51 shows the range 16 to 32 - if spimm[4]=0 the stack adjustment is 16, if spimm[4]=1 the stack adjustment is 32

I've added a note in the spec about this - is it clear now?

tariqkurd-repo, 2021-06-22:
Hi @Xinlong-WU let me know if this needs reopening

closed 2021-06-22 by tariqkurd-repo