riscvarchive/riscv-code-size-reduction #75: a suggestion about the grammar of the reglist

Marcythm, 2021-06-20:
Since `alist` is completely determined by `slist`, and there is a bijective correspondence between `slist` and its length, why not use the length of `slist` instead of the two operands `slist` and `alist` (or only `slist` in `pop/popret`)?
And this also moves the constraints between `slist` and `alist` from the semantic level to the syntactic level, eliminating the case where `slist` and `alist` do not correspond.
e.g. `push 3, -128` instead of `push {ra, s0-s1}, {a0-a1}, -128` where the only valid `slist` with length 3 is `{ra, s0-s1}`, and the corresponding `alist` is `{a0-a1}`.

Marcythm, 2021-06-20:
and considering the 16-bit encodings, we can change the meaning of the number `n` representing a `slist` from "a `slist` with length `n`" to "a `slist` with the `n`th-smallest length (among all valid `slist`)"

tariqkurd-repo, 2021-06-20:
Hi @Marcythm thanks for your suggestion. 
I think it's clearer if we know exactly which registers are pushed/popped instead of only a number - especially in the case of the 16-bit encodings where "c.push 6" doesn't mean push 6 registers in your example.

I agree that the syntax has a lot of redundancy for `alist`, but I think it makes is very clear to read.

`c.push {ra, s0-s7}, {a0-a3}, -128`

I know it will push `{ra, s0-s7}` and it will move `{a0-a3}`, without needing to refer to the ISA manual to find out which registers are pushed, and which registers are moved.

The current proposal matches Huawei's push/pop implementation (which doesn't include `alist`) so we already have a toolchain which supports:

`c.push {ra, s0-s7}, -128`

maybe `alist` could be simplified to be a number only, for example:

`c.push {ra, s0-s7}, {4}, -128`

would that make much difference?

Tariq

Marcythm, 2021-06-20:
Hmm, the biggest problem we encountered is the redundancy of the `alist` operand. i.e. `c.push` has 3 operands but only 2 are used when encoding.

> but I think it makes is very clear to read.

Yes, I admit it. What I'm trying to do is to find the balance between intuitiveness and difficulty of implementation.

> maybe `alist` could be simplified to be a number only

yes it does make the implementation easier to some extent, but considering it’s not intuitive in this way, I think it's unworthy of changing it.

tovine, 2021-06-20:
Maybe a middle ground where only the parts used in the encoding are _required_, but when viewing disassembly the whole redundant syntax is used for ease of reading?

Marcythm, 2021-06-20:
> Maybe a middle ground where only the parts used in the encoding are _required_, but when viewing disassembly the whole redundant syntax is used for ease of reading?

hmmm, do you mean use `c.push rlist, spimm` for assembler, and when disassembling output the whole syntax `c.push slist, alist, spimm`?

tovine, 2021-06-20:
Something like that yes, obviously also supporting the verbose syntax including `alist` for assembler as well (if provided).

tariqkurd-repo, 2021-06-21:
I think it's simpler to include all of it - although I agree that `alist` for c.push is redundant.
These instructions will be generated by the compiler - not by assembly programmers in general - so I think keeping the full syntax isn't a problem.

tariqkurd-repo, 2021-06-22:
let me know if this needs reopening

closed 2021-06-22 by tariqkurd-repo