riscvarchive/riscv-code-size-reduction #76: Clarification of 8.5.2. Exceptions for jump table mode

silabs-robin, 2021-06-21:
Hi. I find the following part troublesome to interpret with certainty, and I also suspect there might be a problem with it.

[image]

https://github.com/riscv/riscv-code-size-reduction/blob/master/ISA%20proposals/Huawei/Zce_spec.adoc#852-exceptions-for-jump-table-mode

## Problem 1

Where is the exception happening? I don't think the text is clear enough about this.
When executing a table jump instruction, there are essentially three places that might go wrong: 1) the tblj itself excepts (e.g. because it was blocked by pma), 2) the table lookup excepts, 3) the target address excepts.
I think the text in "8.5.2. Exceptions for jump table mode" talks about case 2, where an exception happens upon doing the lookup. But I cannot clearly rule out that we are talking about case 3, based on how the spec is written (although then `mepc` should not be the tblj).

## Problem 2

If the paragraph describes a lookup having an exception, how can `mtval` be _"set to the address in the jump table entry"_?
It could happen that the value in the lookup table could not be obtained, and then one can't store it in `mtval`.
Should it be "the address **of**" the entry?

tariqkurd-repo, 2021-06-21:
Hi @silabs-robin ,

Yes there are two I-fetches required, and either can cause a fault. It should be clear *which* caused the fault.

If the jump table entry can't be accessed, then `mtval` should be set to the address of the jump table entry.
If the PC in the jump table entry, i.e. the final target can't be accessed then `mtval` should be set to that.

If the `tblj*` iself can't be executed, then it's standard RISCV behaviour for trying execute an instruction without permission, and there is no more detail to add here.

"`MEPC` is set to the PC of the `C.TBLJ*` instruction, and `MTVAL` is set to the address which caused the fault, which is either the address of the jump table entry or the contents of the jump table entry."

is that better?

tariqkurd-repo, 2021-06-21:
please review my spec edit, I think it's specified much more clearly now.

silabs-robin, 2021-06-22:
Hi @tariqkurd-repo.

> If the jump table entry can't be accessed, then mtval should be set to the address of the jump table entry.

This part is nice and helps clear things up.
Before, it said _"the address **in** the jump table entry"_, and I think that would be impossible since it would mean the address of the target but the exception had just denied access to that information (for the case of fetching the table excepting).
Saying _"of"_ would make more sense to me at least.

> In both cases, `MEPC` is set to the PC of the `C.TBLJ*` instruction

This part is a bit beyond my knowledge, but I vaguely suspect that it is wrong.
If _"the second [memory access] is from dereferencing the jump table entry and fetching the target address"_, why should MEPC be the tblj?
When executing a normal JAL and the target excepts when trying to fetch it, isn't that exception associated only with the target attempted instruction and not with JAL itself? If that is true, then the same should hold for tblj?

tariqkurd-repo, 2021-06-23:
please reopen the issue if you have more questions.

silabs-robin, 2021-07-20:
Hi, Tariq. I am coming back to this issue now, as it became relevant again and I have doubts about the current spec.

> The first memory access is to the jump table, and the second is from dereferencing the jump table entry and fetching the target address.
> [...]
> In both cases, MEPC is set to the PC of the C.TBLJ* instruction

This part seems wrong to me. It seems to be different from how normal jumps set MEPC, and hence break the convention.
I can give some example to illustrate what I mean. If I have made some mistake, please call me out on it.

1. If you run an assembly program that is e.g. "nop" followed by ".word 0x0" (illegal instr), what should mepc be when it reaches an exception?
It should be the address of the 0x0 illegal instruction.

2. If the program is "j" with a target that is "0x0", what should mepc be?
Address of the 0x0 illegal instruction (the jump did its job just fine).

3. If the program is "tblj" with target "0x0"?
I would expect that MEPC is not the tblj.

4. Program is "tblj" with target marked as not executable by PMA.
I would expect that MEPC is not the tblj.

Regards Robin.

silabs-robin, 2021-07-20:
@tariqkurd 

> please reopen the issue if you have more questions.

I don't see that I have permissions to reopen the issue...

tariqkurd-repo, 2021-07-20:
reopened

tariqkurd-repo, 2021-07-20:
Hi @silabs-robin the problem here is being able to resume execution. For example if there's a TLB fault on the jump table entry.
If you set MEPC to the jump table entry itself, then the code isn't restartable, because you can't resume directly to the entry, only to the instruction which called it. The fault handler needs to know the restart PC (MEPC) and the address which caused the fault (MTVAL).

MEPC is always the C.TBLJ* instruction, because that has not completed until the table has been successfuly dereferenced.
MTVAL is  the address which caused the fault, which can be either address (table jump entry or destination function).

and of course both can cause faults, so the fault handler can be called twice with MPEC=C.TBLJ* for the same instruction as *both* of them missed in the TLB.

In your scheme I don't think the code is restartable.

Silabs-ArjanB, 2021-07-20:
Hi @tariqkurd-repo If there is an error associated with the final address (the final jump target address), why can't a replay simply be done by the CPU just performing a regular jump to that final target address (which would be in the mepc). We are not arguing about the mepc value for an error during the table access itself (there we agree it should be the address of the table jump instruction itself). Doing that for the 2nd possible fault as well would really complicate the pipeline we expect.

tariqkurd-repo, 2021-07-20:
Hi @Silabs-ArjanB how would the handler know that the exception was caused by the jump table? There's nothing in the context which represents this, except that the table address is > TBLJALVEC, but the table doesn't have to extend to all 256 entries and so there could be normal code in that range.

We would have to add a new MCAUSE value - "table jump access fault" for example.

If we did this then the exception handler and debug handler would need to change - it needs to just work with the existing  software, otherwise recompiling with Zce will mean that existing software systems stop working. 

For instruction access fault and instruction page fault, and anything else which matches on a fetch address (e.g. breakpoint trigger) the fault can be caused by either address, and must be restartable.

I don't know how you are building your pipeline, but essentially you need to keep a copy of the C.TBLJ* PC for fault reporting if the fault is caused by dereferencing the table entry, and then everything keeps working and we don't have to change any software.

Silabs-ArjanB, 2021-07-20:
Hi @tariqkurd-repo,

> For instruction access fault and instruction page fault, and anything else which matches on a fetch address (e.g. breakpoint trigger) the fault can be caused by either address, and must be restartable.

I am not sure where the root of my misunderstanding is. I fully agree with the quote here above. To me that implies that if we have a PMA error on the fetch address which is the final address of the table jump, that we both seem to agree that the fault can be caused by either address (so also the final target address). If we would put that branch target address in the mepc, then it can be restarted by simply jumping there, The exception handler would indeed not know anything about the table jump, but I don't see why it would have to know that either (e.g. if a regular jump led to a fetch address error at its target address, the exception handler would also not be made aware of the regular jump).

Maybe my misunderstanding is because you use 'TLB' in your example and we don't have a TLB in our core, so I am not sure how its related errors are typically implemented. Is it the 'TLB' part that is essential in your example?

silabs-robin, 2021-07-20:
I think it boils down to answering these:
1. Upon regular jump to a target address which causes an exception, should mepc be the target address?
2. Should tblj behave any different from a regular jump in this regard?
3. Does the Zce spec state that tblj must behave any different from a regular jump in this regard?

tariqkurd-repo, 2021-07-20:
Hi, I don't think the nature of the fault is important, TLB is a good example as any address (I/D) can cause it.

Given this example:

00000050: tbljal #20
00000100: table entry = 00008000
00008000: <function>

*00000100 has a TLB fault*

MEPC   = 00000050
MTVAL = 00000100

we agree on this one - and execution resumes at tbljal #20 

*00008000 has a TLB fault*

so here I suggested that 
MEPC   = 00000050
MTVAL = 00000100

which is causing you problems, so are suggesting that:

MEPC   = 00000100
MTVAL = 00000100

?

I thought you were suggesting that MEPC = 00000050, which means that the handler can't resume. However if MEPC points to the function then it's fine, it can resume execution. So this *is* different to a normal jump which 

MEPC = PC of the jump
MTVAL = PC of the target

is that what you wanted?

Tariq

Silabs-ArjanB, 2021-07-21:
00008000 has a TLB fault

> 
> so here I suggested that
> MEPC = 00000050
> MTVAL = 00000100
> 
> which is causing you problems, so are you suggesting that:
> 
> MEPC = 00000100
> MTVAL = 00000100
> 
> ?

No. If 00008000 has a TLB fault, then our proposal is to do:

MEPC = 00008000
MTVAL = 00008000

> How do normal jumps behave? I was thinking that it's this:
> 
> MEPC = PC of the jump
> MTVAL = PC of the target

If the error is related to the target address, then we do the following:

MEPC = PC of the target
MTVAL = PC of the target

I can't really find any requirement on this in the RISC-V specifications, but using 'PC of the target' makes the implementation much easier as the exception information can then flow together with the 'invalid fetched data' through the prefetcher and pipeline.

tariqkurd-repo, 2021-07-21:
ok - please tell me if the new text is ok

closed 2021-07-21 by tariqkurd-repo

silabs-robin, 2021-07-21:
@tariqkurd-repo The new text appears to be aligned with my understanding and expectations.
There should probably be a more rigorous review of the spec at some point but at least this part about tblj exceptions looks good to me.

tariqkurd-repo, 2021-07-21:
Hi @silabs-robin - yes - this part definitely needs thorough review and it's likely that it will need some small changes later, but I'm glad that I've solved your current problem, and so you can continue with your implementation.