riscvarchive/riscv-code-size-reduction #79: Code size analysis script pushpop analysis with msave-restore

Silabs-ArjanB, 2021-06-25:
Hello,

Thank you very much for making the code size analysis script available. In https://github.com/riscv/riscv-code-size-reduction/tree/master/benchmarks it is clearly stated that the 'pushpop' action of the script 'only works correctly with files compiled without msave-restore'.

Of course, when going after lowest possible code size, we would like to use msave-restore. Will the script be updated to make analysis of pushpop effectiveness possible when msave-restore is being used? If not, will the intended prototype compiler allow for benchmarking pushpop with msave-restore enabled?

tariqkurd-repo, 2021-06-25:
We report results relative to using -msave-restore, otherwise the push/pop results are ridiculously high.
@abukharmeh can tell you how that works - and maybe put something in the documentation about it.

Silabs-ArjanB, 2021-06-25:
I forgot to mention: The script will crash if it detects that a binary is compiled with the msave-restore option and you try to check the benefits of pushpop. 

This aligns with the following comment in the script documentation, but is very unfortunate:

[image]

abukharmeh, 2021-06-25:
Hi Arjan, 

The script cannot effectively detect stores and loads when they are transformed into save restore calls, so it requires files that are compiled without it. Like @tariqkurd-repo  said, we compare the results to files compiled with m save restore, this is performed using "pushpop_calc" file, this script takes two input CSV files one with save restore and one without and it would report the results relative to save restore.

The analysis script assert the non existence of save restore routines to make sure its matching criteria is valid otherwise it exit like you said. 

Locally I have a method that emulate save restore routines as well, so both push pop and save restore are emulated and both work on files compiled without save restore, I will try to get that to public as soon I can. 

Kind regards,
Ibrahim

kjetilos, 2021-06-25:
Hi Ibrahim,

What options do you use in the hca script to create the necessary csv files to run the pushpop_calc analysis? When I'm trying to generate csv files I either get empty files or files that the script is unable to parse.

abukharmeh, 2021-06-25:
That script is really not part of the analysis script, and its just a helper script I used to help me calculate the results of pushpop relative to save restore.  I decided to push it along just in case somebody find it useful. 

So pushpop in the version of HCA on Github can be called hca -a pushpop,both size -f afile.elf -r ~/ , the both option passed to pushpop action is to allow it to use both 16 and 32 bit encodings of pushpop instructions.  You would need to run this on a file/ files that **do not contain save restore** routines so its actually able to calculate the savings. 
 *Sorry, I need to add this option to the documentation as I just realised its not documented*

After that, you need to run hca on the same files with the same names **compiled with save restore**  just to calculate the instruction record size 
e.g hca -a size -f afile.elf -r~/  (Benchmarking reference)

After that if your run pushpop_calc, it would find both columns and combine them and calculate the relative savings. Sorry if this is a little bit convoluted but its the only way I found to estimate push pop relatively accurately.

Silabs-ArjanB, 2021-06-28:
Hello Ibrahim, thank you for the detailed explanation. We will give that a go. Best regards, Arjan

closed 2021-06-28 by Silabs-ArjanB