riscvarchive/riscv-code-size-reduction #81: Code size analysis script longaddress ignores section argument

kjetilos, 2021-06-25:
When running the hca script on an elf file with the "longaddress" action the output tells me that I have a number of long address jumps in my .bss and .debug_info section. I tried to limit the script to only search in the .text section by supplying a "-j .text" argument, but it looks like the script still searches in all sections.

I also get a number of hits in the "unknown" section. What does this mean?

tariqkurd-repo, 2021-06-25:
@abukharmeh

abukharmeh, 2021-06-25:
Hi Kjetilos, 

longaddress action is used to search for load and store words instructions that needed some other instruction before it to construct the load/store address. 

e.g: 
[image]

I created it as an adhoc action when we were evaluating LWGP/SWGP instructions (load/store instructions with longer immediate range similar to lwsp), then decided to keep it as I used it multiple times for other purposes. The idea of that action is that it will find all those lw/sw instructions that required longer immediate, and it will report to you from which section this load/ store is happening. 

When this action is reporting unknown, it means that the load/ stores are to somewhere outside sections reported by "readelf" , my gut feeling is that they are writes to a memory mapped peripheral !! 

When you pass -j + section name, the instruction record would contain parsed disassembly only from that section (This is useful when we have multiple sections of executable code e.g bootloader and firmware as two separate sections in the same elf file), and I added that as sometimes some elf files would contain some precompiled code placed into final elf file, so changing compilation parameters wont change such sections, so in some places I wanted to exclude such sections from analysis. 

Kind regards, 
Ibrahim

kjetilos, 2021-06-25:
Thanks Ibrahim,

So in your example there is a load of a global variable (@0xF00100), what would this be reported as? Would it be reported as unknown or as a variable in .bss perhaps.

abukharmeh, 2021-06-25:
That file contains multiple BSS sections, and it would report it as part of .shared_rom_bss. The results from this action really contain the number of times it needed to construct a long address to access something from each section.

closed 2021-07-05 by kjetilos