riscvarchive/riscv-code-size-reduction #82: assembler syntax of C.*.E

Xinlong-WU, 2021-06-29:
in **9.3.2. 16-bit encoding assembler syntax**, it describes the assembler syntax of `c.push.e`, `c.pop.e` and `c.popret.e`.
[image]

therefore I have the following issues

1.  for `c.push.e`, how does `{<areg_list>}` works? cause in _Table 48_ just use `areg_list` only in `C.PUSH`, should I process `c.push.e` in the same way?
2.  for `c.pop.e`, What is the purpose of the empty brackets? it seems useless in encoding, `c.pop` also haven't this parameter.

tariqkurd-repo, 2021-06-29:
Hi - there was a mistake: table 48 should have said C.PUSH.E only
The {} for C.POP[.E] and C.POPRET[.E] are to specify the return value, and to have consistent assembler syntax with POP[.E] and POPRET[.E].

It is true that there is no valid return value for C.POP[.E]

Maybe it's better if I simplify the C.POP[.E] syntax so it doesn't have the {}, field, but this may effect the other tools which are under development.

What do you think?

Xinlong-WU, 2021-06-29:
ok, also another question.

it said `C.POP.E (RV32 only)` in Table 46.
but why there still a column for C.POP.E in Table 53?

also, Table 48 describes the valid registers for `rlist2`, only `ra, s0-s2`, `ra, s0-s3` and `ra, s0-s4`.
does it mean that only three rows_(`{ra, s0-s2}`,`{ra, s0-s3}`,`{ra, s0-s4}`)_ will have a valid number for C.POP.E in Table 51?

tariqkurd-repo, 2021-06-29:
ok - thanks for finding that! I've updated the tables. In (e.g.) Table 51 I've tried to make it clear when the .E instruction is a pseudo-instruction.

is it ok now?

closed 2021-07-01 by Xinlong-WU