riscvarchive/riscv-code-size-reduction #83: Binary instructions conflicts on C.PUSH and C.PUSH.E

Xinlong-WU, 2021-07-01:
|instructions|binary code|
|---|---|
|C.PUSH.E encodings |`1000 1111 _1__ __00`|
|`c.push.e {ra,s0-s2}, {a0-a2}, -96`|`1000 1111 0101 0000`|
|C.PUSH encodings |`1000 11__ _10_ __00`|
|`c.push {ra,s0-s3}, {a0-a3}, -32`|`1000 1111 0101 0000`|

In this table, you will find that two instructions `c.push.e {ra,s0-s2}, {a0-a2}, -96` and `c.push {ra,s0-s3}, {a0-a3}, -32` has same binary encoding. 
what should we do when we meet encoding, Do we need to prioritise the use of C.push.e?

tovine, 2021-07-01:
One is for implementations with RV32E and the other for RV32I base ISA

Xinlong-WU, 2021-07-01:
OK thanks, but actually I'm not very familiar with the current status of the E extension, can you give me a link to the E extension related repository?

tovine, 2021-07-01:
The E base is already defined in the unprivileged ISA manual, but work is ongoing on a new EABI - that can be found here: https://github.com/riscv/riscv-eabi-spec/blob/master/EABI.adoc

tariqkurd-repo, 2021-07-01:
Hi @Xinlong-WU , RV32E is a different base architecture supporting 16 X (integer) registers, not 32.
The normal RISC-V ABI is the UABI, which uses all 32 registers. There is also the EABI proposal which is required for RV32E, and is also an option for RV32I as it reduces the number of registers which need to be saved on a context switch (see the link which @tovine posted above).

C.PUSH.E etc. should only be selected if compiling for RV32E or compiling for RV32I + EABI.

But - that's useful information but it's not the problem here. The problem is this:

100 | 011 | `spimm0_5[6:4]` | 1 | 0 | rlist3 | 00 | C.PUSH
100 | 011 | `11 `| spimm0_5[4] | 1 | spimm0_5[6:5] | rlist2 | 00 | C.PUSH.E (RV32 only)

`spimm0_5[6:4]` only has legal values of 0 to 5, the way we distinguish between C.PUSH and C.PUSH.E is

`[9:8] != 2b11 -> C.PUSH`
`[9:8]  = 2b11 -> C.PUSH.E`

is that ok?

Xinlong-WU, 2021-07-02:
I'm still confused about the way of distinguishing between C.PUSH and C.PUSH.E.

for C.PUSH, `[9:8] = spimm0_5[6:5]`

for C.PUSH.E,  `[9:8] = 2b11`

however, spimm = `2b1110 0000` when stack adjustment is `-32`, thus the spimm0_5[6:5] is `2b11` too, same as C.PUSH.E.

tariqkurd-repo, 2021-07-02:
Hi - `spimm` is always positive, it's an unsigned immediate/

https://github.com/riscv/riscv-code-size-reduction/commit/0b0d9745e9eb7144effd2e3379e88d61b876179e#diff-100a61e29d49481166f34a547a568eea417dd88c57acab910bff175d76e680dd

I've added a note about it, and the pseudo-code shows the correct usage.
Maybe it should be called `spuimm` with `u` for `unsigned`

is it clear now?

Xinlong-WU, 2021-07-02:
good, thanks

Xinlong-WU, 2021-07-02:
good, thanks

Xinlong-WU, 2021-07-03:
hi, I reopen this issue.
you have mentioned "`spimm0_5[6:4]` only has legal values of 0 to 5", but I found an issue again.
for `c.popret`
Table 53 describes the range of stack adjustment for `c.popret`.
following is `spimm0_5[6:4]` and its stack adjustment when the range is 16 to 96
[image]
the first column is the binary code of `spimm0_5[6:4]` and the second is stack adjustment value.

we find `spimm0_5[6:4]` is out of range when stack adjustment is 96.
96 = `0b 0110 0000`, the spimm0_5[6:4] sould be `0b110` but it is 6.

I find that `B000` has not been used in this case, so do we need to encode 96 as `B000`？

But honestly, I don't really recommend this because I'm sure there are many more similar situations and it would be an overly complicated thing for us to do if each one was handled specifically

tariqkurd-repo, 2021-07-04:
Hi - it's simpler than that.
If the registers take <= 16 bytes of memory (e.g. the top row of table 53) therefore the minimum stack adjustment is 16. spimm gives *extra* stack adjustment, so it is encoded as follows:

spimm=0 -> 16
spimm=1 -> 32
spimm=2 -> 48
spimm=3 -> 64
spimm=4 -> 80
spimm=5 -> 96

the total stack adjustment is always the minimum required for all of the registers, rounded up to 16-byte blocks, and then add 16*spimm.

so in tables 51, 52, 53, 64 the minimum value always has spimm=0 and is only enough to cover all of the registers.

I hope that's clear not. The 0-5 encoding scheme is a bit unusual, but it saves encoding space.
Let me know if you have more problems!

Tariq

Xinlong-WU, 2021-07-04:
ok,good , that helps a lot.

You can give a formula for getting Spimm0_5 from the stack adjustment, 
e.g. in this case, it is `Spimm0_5  = (stack adjustment - 16) / 16`

closed 2021-07-04 by Xinlong-WU

tariqkurd-repo, 2021-07-05:
Hi,

It's not always 16 - it needs to be enough bytes to cover of the registers, rounded up to the nearest 16-bytes 

`spimm0_5 = (stack adjustment - enough_16_byte_blocks_to_cvoer_the_registers) / 16`

It's described here: https://github.com/riscv/riscv-code-size-reduction/blob/master/ISA%20proposals/Huawei/Zce_spec.adoc#971-spimm-encoding-field

there's also this note under the tables

`Note | The ranges in Table 51 and Table 52 show the total stack adjustment range, starting from spimm=0 up to the maximum value of spimm for each encoding.`

so I think the document is ok?!

Xinlong-WU, 2021-07-05:
I know that is not always 16. I mean we use `Spimm0_5 = (stack adjustment - 16) / 16` when we process the instruction `c.popret {ra}, {}, stack adjustment`. 

I just consider that might be easier for understanding if we give an example.

tariqkurd-repo, 2021-07-05:
https://github.com/riscv/riscv-code-size-reduction/commit/ac5bc9b449a1ecce9bbba479e71bbf702f25bfc6

I hope that improves the explanation