riscvarchive/riscv-code-size-reduction #93: hca: Crash when analysing elf file with all Zce actions applied

kjetilos, 2021-07-08:
This was seen when analyzing an elf file using the following command line:

```
hca -a c_sext_b c_sext_h c_zext_b c_zext_h c_zext_w beqi bnei muli c_mul c_not c_neg c_tblj c_tbljal c_tbljalm c_mva01s07 pushpop pushpop,both c_sb c_sh c_lhu c_lbu c_lh c_lb sp,lbu sp,lhu sp,sb sp,sh size -f file.out -j .text
```

Here is the python trace.

```
Traceback (most recent call last):
  File "tools/HCA/hca", line 401, in <module>
    main()
  File "tools/HCA/hca", line 204, in main
    (push,pop) = program.find_push_pop(compressed)
  File "C:\work\repos\riscv_benchmark\tools\HCA\elf.py", line 1016, in find_push_pop
    if ("ra" in Wasted_moves and "c." not in inst):
UnboundLocalError: local variable 'Wasted_moves' referenced before assignment
```
It looks related to this code:

https://github.com/riscv/riscv-code-size-reduction/blob/16585d4eb2fdec79055caf708aeba314792f9187/benchmarks/HCA/elf.py#L1016

abukharmeh, 2021-07-08:
I could not replicate this by just running the exact same command, I would assume its triggered somehow by the input elf file, is the elf file you are using something you can share ? 

Otherwise maybe I can try to guess how its generated !

abukharmeh, 2021-07-08:
Actually I managed to trigger it, studying it to find the exact cause.

kjetilos, 2021-07-09:
Hi,

Unfortunately I cannot share the elf file. The PR looks good to me, I will apply this fix locally to be able to analyze my elf file. Thanks for the quick feedback.

abukharmeh, 2021-07-09:
You are welcome, please let us know if this fixes your issue, so we can merge the PR and close the issue.

PR #94 (2021-07-12): Fixes a typo in script
This is already fixed in upcoming patch, but its completely wrong here and should be amended immediately, fixes #93
closed 2021-07-12 by tariqkurd-repo