riscvarchive/riscv-code-size-reduction #96: SP relative non-word load/store

Silabs-ArjanB, 2021-07-13:
Hello,

We ran the measurement script including sp,lbu sp,lhu sp,sb sp,sh and these instructions seem quite useful in our benchmark:

Zceb extensions
---------------
c_sb        0.03%
c_sh        0.02%
c_lhu       0.02%
c_lbu       0.40%
c_lh        0.00%
c_lb        0.00%
sp,lbu      0.16%
sp,lhu      0.00%
sp,sb       0.22%
sp,sh       0.05%

I do however not see these instructions actually documented. Are these instructions not (longer) part of Zce? If not, why were they excluded?

tariqkurd-repo, 2021-07-14:
Hi @Silabs-ArjanB , those instructions were considered.

https://github.com/riscv/riscv-code-size-reduction/blob/master/ISA%20proposals/Huawei/Zce_spec.adoc#c-lbusp-c-lhusp-c-sbsp-c-shsp

In general the benefit was low, but the encoding space they need - 7 bits / 128 code points is a lot.

However - it is simple to check the results again on the benchmark suite, @abukharmeh can you do that before the code-size meeting next week?

abukharmeh, 2021-07-14:
Yes in theory, it should be pretty simple to regenerate them and add them back to results sheet.

Silabs-ArjanB, 2021-07-14:
Hi @tariqkurd-repo , thanks for your answer (I had missed that section of the spec, sorry for that). I agree that these instructions are expensive, but so are of course the more generic C.LBU, C.LHU, C.LB, C.LH, C.SB, C.SH instructions (each of them use 8 times as many code points as they have rs1' programmable). Do you think that for example C.SB has 8 times better savings than C.SBSP? If not, would it make sense to use a new 3-bit rs format (let's call it rs1'' in this thread) for the C.LBU, C.LHU, C.LB, C.LH, C.SB, C.SH instructions with the rs1'' semantics being the same as the rs1' semantics except that for example rs1'' = 0x7 means 'sp' as opposed to 'x15'.

Of course such a new rs1'' encoding will lead to a less orthogonal instruction encoding, but I don't see why that would be a concern. I would expect (but did not collect data to proof) that 'sp' is more useful/common as address base register for loads and stores than the least popular register out of x8-x15.

tariqkurd-repo, 2021-07-14:
Oh I see - so instead of the base register being x8-x15 it would be x8-x14, sp.

I had a quite look at the Huawei IoT code, these are the most popular base registers for LB, LBU, LH, LHU, SB, SH

  12090 s0
   7213 a5
   6038 sp  <--- not available, but we want it
   5472 a0
   4536 s1
   2475 a4
   2040 s2 <--- not available
   1860 a1
   1325 s3 <--- not available
    966 a3
    601 a2

out of 48003, so s0 > 25%
replacing a5 with sp would clearly make the situation worse.

Note that we already have a special register mapping for C.MVA01S07:

`//000 (s0) -> 01000 (x8)`
`//001 (s1) -> 01001 (x9)`
`//010 (s2) -> 10010 (x18)`
`//011 (s3) -> 10011 (x19)`
`// ...         ...`
`//111 (s7) -> 10111 (x23)`

`//verilog syntax for concatenation of bits`
`xreg[4:0] = {sreg[2:1]>0,sreg[2:1]==0,sreg[2:0]};`

so it's only adding 2 upper bits.

to do a similar trick with c.lb etc. would be quite irregular, a0 has the correct bits [2:0] (010) but we really wouldn't want to replace that one.

but it's a good point that we can save encoding space with a better register mapping, something to consider.

Silabs-ArjanB, 2021-07-14:
Replacing x15 by x2 (sp) was just an example and based on your numbers above a bad example. For the Huawei IoT code it would be best to replace a2 by x2 (sp). Does the irregular encoding really matter? (I realize that it might make the (pre-)decoder slower, but that would be 1-2 gates at most I would estimate). Maybe we should ask a compiler expert to comment on it and indicate which base registers are 'easy to use' and 'expected to be useful'  for the compiler and whether it matters if these are not consecutive.

tariqkurd-repo, 2021-07-14:
Currently rs1[2:0] always come directly from encoding[9:7] for 16-bit instructions. 
rs1[4:3] vary - as they can be zero or from encoding[11:10].
C.MVA01S07 adds another case, changing the value of rs1[4:3].

I don't think we should add logic onto  rs1[2:0] if we can help it.

Having said that - the C.LBU encodings reuse D-extension encodings, so they will not be used for high performance machines. This means that maybe we can do something in this area, but I'm nervous of breaking the paradigm of rs1[2:0] always being directly from the encoding - meaning that x10/a0 is the obvious choice but is used to often to replace it with SP (in the limited example above).

I'm not sure what to do about this one. Features which break RISC-V paradigms are harder to get ratified (for obvious reasons) but the code-size saving in this case should be pretty good for no increase in encoding space.

I'll think about what to do.... and we can discuss it in the meeting next week

tariqkurd-repo, 2021-07-14:
@abukharmeh maybe we can do some modelling work on this before tuesday

abukharmeh, 2021-07-14:
More data across the whole benchmark suite 
[image]

abukharmeh, 2021-07-14:
[image]

I wrote a simple AWK script to estimate this across the whole benchmark suite and it looks like a2->sp is indeed the best change !

Silabs-ArjanB, 2021-07-15:
Hi @tariqkurd-repo , @abukharmeh 

Ibrahim, thank you very much for doing the above analysis. In my opinion that is indeed enough input to the discussion in next week's meeting.

>> Currently rs1[2:0] always come directly from encoding[9:7] for 16-bit instructions.
>> …
>> I don't think we should add logic onto rs1[2:0] if we can help it.

I assume the above (i.e. the 'rs1[2:0] coming directly from encoding[9:7]')  is true for Zce, but it is not true for the C extension in general because we already have instructions as for example C.LWSP that add additional logic on the paths to rs1[2:0] in order to force rs1 to point to 'sp'. Although I agree that we should avoid adding logic onto rs1[] in general, it is not obvious to me that allowing other compressed instructions to force 'sp' to be used as rs1 would make timing worse.

Based on above results I would actually even go further and propose: a2 -> sp and a3 -> s2 for all of the C.LBU, C.LHU, C.LB, C.LH, C.SB, C.SH instructions.

I realize this thread is going a bit off topic. As you already answered my original question, please feel free to close this issue (or to keep it open as a reminder for the rs1' format discussion).

tariqkurd-repo, 2021-07-15:
Hi @Silabs-ArjanB - yes that's a good point C.LWSP already forces SP onto rs1, so another instruction which does the same should be fine.
For the actual register mapping - we definitely need some compiler expert input - maybe it's time to ask the mailing list.

jnk0le, 2021-07-15:
There is one significant prolem with deciding that `a2` or any other register should be a second class citizen. Compilers can change schedulers at any time, especially that they don't try to target the zce added instructions right now (or even all compressed instructions at all).

Also the alternative to byte/half acces directly through sp could be `c.addi4spn` and then acces it by `c.lb`/`c.lh`/etc. (depending on what are typical patterns where byte/half acces through sp is used)

jnk0le, 2021-07-15:
Another way is to implement `C.L*SP` and `C.S*SP` over `C.FLWSP`, `C.FSWSP` for RV32 (Zfinx) only.

tariqkurd-repo, 2021-07-15:
No - we're not having standalone `C.L*SP` etc. instructions.
The compiler will choose a register to load or store. *if* it is in the 8 register set *then* you will get the 16-bit encoding. This is already how it works. by including *sp* instead of (e.g.) *a2* in the set then it is _more likely_ that the 16-bit encoding will be used.

jnk0le, 2021-07-15:
>No - we're not having standalone C.L*SP etc. instructions.

but it can be implemented similarly to how is currently zceb, as RV32 only addition.

>The compiler will choose a register to load or store. if it is in the 8 register set then you will get the 16-bit encoding. This is already how it works. by including sp instead of (e.g.) a2 in the set then it is more likely that the 16-bit encoding will be used.

The problem is that, when compilers start to optimize for rvc/zce it might turn out that a2 was not a best choice.

Silabs-ArjanB, 2021-07-16:
>> No - we're not having standalone C.L*SP etc. instructions.

Based on above benchmarking results the compressed SP relative non-word load/stores match 7669 times whereas the compressed non-SP relative non-word load/stores match 46485 times. As the compressed SP relative non-word load/stores use only 1/8 of the code points, I would argue that they are actually more justified than the general compressed non-SP relative non-word load/stores. (Don't get me wrong here; I would like to have support for both.)

On the mailing list the discussion is ongoing on whether the alternative rs1'' format is worth pursuing (with the discussion focusing on possible negative impact on critical timing paths). If that discussion converges to not supporting an alternative rs1'' format, then I would like to propose reconsidering adding standalone SP based non-word load/store instructions (in the space currently aimed for by C.DECBNEZ (and incompatible with 'zcd' and 'zc128' (i.e. compressed quadrant 2, inst[15:13] =101, inst[1:0] = 10) as these instructions are already SP based). 

All these proposed SP relative instructions combined would use 768 code points, using only part of the 2048 code point space currently reserved for C.FSDSP.

tariqkurd-repo, 2021-07-16:
Don't forget we have the full set of C.LB, C.LBU, C.LH, C.LHU, C.SB, C.SH.
Looking at the stats, yes SP is always at least 1/8 of the total benefit, so we can argue that it's as valuable, or more valuable than any other register.

Current C.DECBNEZ has 11 programmable bits 0 - which is a lot and is sufficient to cover C.LBSP etc.
It will still mean spending a lot more encoding space compared to swapping an unpopular register for SP in the rs1'' format - but - it's encoding space shared with D so I don't think it's as precious as encoding space which is currently unused - so maybe.

it would certainly be a cleaner solution.

tariqkurd-repo, 2021-07-16:
also see this thread:

https://lists.riscv.org/g/tech-code-size/topic/16_bit_load_store_byte_half/84221184?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,84221184