riscvarchive/riscv-code-size-reduction #101: The definition of the instruction zext.b is located in which extension

Xinlong-WU, 2021-08-03:
We had some problems implementing the instruction optimization of `zext.b -> c.zext.b`, so we needed more details from the definition of `zext.b`, but we didn't know which extension it was defined in.

Could you please tell me which extension it belongs to or give us a link to it?

tariqkurd-repo, 2021-08-03:
Hi - zext.b, zext.h, sext.b, sext.h are in the B-extension.
We don't define 32-bit versions of them in Zce, there's no requirement to have 32-bit versions of all 16-bit instructions.
C.MVA01S07 and C.TBLJAL for example don't have 32-bit versions.

Tariq

Xinlong-WU, 2021-08-03:
ok, thanks

closed 2021-08-03 by Xinlong-WU

jim-wilson, 2021-08-05:
zext.b is an alias for an andi instruction so it is an I instruction.  The others are in the zbb or zba extensions.  If you don't have zbb or zba, then they are macros that expand to a pair of shifts.

Xinlong-WU, 2021-08-06:
so did you mean we need to process instruction like `andi a0, a0, 255` when we try to optimise `zext.b`？

thanks, that's help a lot.

jim-wilson, 2021-08-06:
andi and zext.b are the same instruction.  Whether you need to care about both depends on exactly what you are doing, e.g. are you looking assembly code, how was the assembly code written, are you looking at disassembly, how are you disassembling, etc.